midilab/uClock #2: Teensy support?

grayxr, 2020-10-20:
Looks like some of the core AVR timer code won't work with the Teensy boards, per Paul's answer here: https://forum.pjrc.com/threads/28490-teensy-3-1-intterupts-error-TCCR1A-was-not-declared-in-this-scope

I assume this library would have to be heavily refactored to support the Teensy boards?

midilab, 2020-10-20:
Teensy's above 2.0 are not AVRs, they're all ARM, so different registrers and interruptions.

You can make use of uClock on Teensy 2.0 that has the same microcontroller as arduino leonardo.

I dont have a teensy, but its on my todo list to order a teensy 3.6 and start playing with to code a uClock support for those teensy boards.

I think theres not much of refactor, just a matter of initialize and control correctly any 16 bits avaliable timmer on teensys ARM boards.

grayxr, 2020-10-20:
Thank you @midilab -- I've been trying to get this timer code to work, but I think my interval calculation is wrong: https://gist.github.com/grayxr/fe7b24c2d737763809a553b77b025c26#file-tclock-cpp-L88

Plus, I'd love to use your library for the Teensy because it looks really usable. I wouldn't mind sending you a Teensy 4.1 to experiment with 😄

midilab, 2020-10-20:
That will be great, send me a email so we can talk about it. :)

midilab, 2020-10-26:
Ok, here it comes the first version of uClock and Teensy arm boards support. I had tested on Teensy LC, but all ARM boards uses the same programming interface for timers, what should be ok to use on any other Teensy.

Please post tests made on other board other than LC to update the compatibility chart of uClock.

https://github.com/midilab/uClock/releases/tag/0.9.0

grayxr, 2020-10-27:
Awesome, thank you!

commit cca6dd87 (2017-12-04): Fix #2 for clock PPQN resolutions of 16, 32 and 96
closed 2020-10-27 by midilab