midilab/uClock #3: What external signal can I sync to?

yentzee, 2020-11-05:
Hi,
is this library meant to feed a midi clock into the arduino and then generate different sync signals out of it or would it be possible to use an audio signal (ie 96 short pulses per quarter) that the arduino generates other sync signals from or does it have to be DC? Sorry for maybe asking dumb questions, I am having sync problems with Ableton and my external gear and being able to use an audiofile for timing reference would be awesome. Can't afford something like a multiclock unfortunately :( 
If this is possible what specifications would the signal have to have?
Thanks for sharing your work :)

midilab, 2020-11-06:
Hi yentzee,

I came up with this library as base for other musical timing projects(specially sequencers).

The library is a generic code to handle clock in BPM notation using a hardware interrupt to generate predictive clock. It also has a phase lock mechanism to handle external clock to generate own internal clock as tigth as the source clock.

For external clock you have a method called clockMe(); expecting a 96ppqn as base clock, but it can be divided, or multiplied via your added code with a 16 microseconds resolution. 

You can grab any kinda of signal you need, just call the clockMe(); when the signal arrives, this library doesnt deals with integration abstraction, its only to use as a clock genarator for projects.

So what you need is to find the correct electronic schematic diagram over the internet for the signal you want to interface with your arduino/teensy. When the signal arrives, triger the clockMe() call and you are good to generate internal clock for a sequencer or to resend this clock to modular gears on euroack or more than 4 midi ports at once. 

I normaly use a lot with ableton live, so i would say that it syncs quite gently as master or slave clock over the USB or uart MIDI.

Rigth now it generates a 96ppqn, 32ppqn and 16ppqn interrupetd outputs, i kinda plan to work on better divider and multiply setup for the clock. just keep in touch for new releases...

yentzee, 2020-11-06:
Thank you so much for your help and for sharing your work.
Gonna start experimenting at the weekend I think.
Best regards from Berlin,

Jens

commit a3dcc0b1 (2017-12-04): Fix #3 for clock PPQN resolutions of 16, 32 and 96
closed 2020-11-10 by midilab