midilab/uClock #6: uClock support for Seeeduino XIAO?

m-r-m-s, 2022-04-25:
Hi @midilab, 

I am considering using the [Seeeduino XIAO](https://wiki.seeedstudio.com/Seeeduino-XIAO/) for a MIDI project and wanted to know if it is supported? It seems possible as the XIAO comparable to the Teensy LC. Perhaps with the 1.0 release it is now supported but I wanted to confirm. I found this [old thread](https://forum.seeedstudio.com/t/interrupts-and-nointerrupts-is-undefined/254943) that may provide some clues. 

XIAO Reference: https://wiki.seeedstudio.com/Seeeduino-XIAO/#features

Also reference here: [https://www.sigmdel.ca/michel/ha/xiao/seeeduino_xiao_01_en.html#timers](https://www.sigmdel.ca/michel/ha/xiao/seeeduino_xiao_01_en.html#timers)

Thank you!

m-r-m-s, 2022-04-26:
Hi @midilab thank you for the reply. I will look further into XIAO's arduino support and get back with you as soon as I know more. Thanks so much!

mightycoco, 2022-04-27:
I have tried to implement support for XIAO_M0 in https://github.com/mightycoco/uClock/commit/ba26ca62cce8941f4b3001303339e5edc5a0a6eb

It's compiling but I haven't tested right now, as I run out of MCUs :/

You need to have the latest SAMD board for Seeeduino installed which should be 7.1.0

m-r-m-s, 2022-04-27:
> I have tried to implement support for XIAO_M0 in [mightycoco@ba26ca6](https://github.com/mightycoco/uClock/commit/ba26ca62cce8941f4b3001303339e5edc5a0a6eb)
> 
> It's compiling but I haven't tested right now, as I run out of MCUs :/
> 
> You need to have the latest SAMD board for Seeeduino installed which should be 7.1.0
The latest version I see for the Xiao is 1.8.2? 

Thank you for doing this @mightycoco ! I tried your fork with a [modified version](https://gist.github.com/m-r-m-s/ed0dba5025ac58d5189581e0a6bd653c) of the @midilab example `TeensyUsbMasterMidiClock` for the Xiao. It does not compile (the verbose error message is below my modified sketch). Short version:
```
Arduino: 1.8.19 (Mac OS X), TD: 1.56, Board: "Seeeduino XIAO, TinyUSB, Off"

WARNING: library uClock-xiao claims to run on avr, arm architecture(s) and may be incompatible with your current board which runs on samd architecture(s).
/Volumes/HDD/wetmath/Documents/Arduino/2022-04-27-uClock-Xiao-sketch_apr27a/2022-04-27-uClock-Xiao-sketch_apr27a.ino: In function 'void setup()':
2022-04-27-uClock-Xiao-sketch_apr27a:58:45: error: invalid conversion from 'void (*)(uint32_t*) {aka void (*)(long unsigned int*)}' to 'void (*)(volatile uint32_t*) {aka void (*)(volatile long unsigned int*)}' [-fpermissive]
   uClock.setClock96PPQNOutput(ClockOut96PPQN);
                                             ^
In file included from /Volumes/HDD/wetmath/Documents/Arduino/2022-04-27-uClock-Xiao-sketch_apr27a/2022-04-27-uClock-Xiao-sketch_apr27a.ino:10:0:
/Volumes/HDD/wetmath/Documents/Arduino/libraries/uClock-xiao/src/uClock.h:120:8: note:   initializing argument 1 of 'void umodular::clock::uClockClass::setClock96PPQNOutput(void (*)(volatile uint32_t*))'
   void setClock96PPQNOutput(void (*callback)(volatile uint32_t * tick)) {
        ^~~~~~~~~~~~~~~~~~~~
exit status 1
invalid conversion from 'void (*)(uint32_t*) {aka void (*)(long unsigned int*)}' to 'void (*)(volatile uint32_t*) {aka void (*)(volatile long unsigned int*)}' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

```
/*
 * Modified version of uClockTeensyUsbMasterMidiClock example for Seeeduino XIAO
 * https://github.com/midilab/uClock/tree/master/examples/TeensyUsbMasterMidiClock
 * 
 * ////////NOTE: MUST USE VERSION 0.10.5 Adafruit_TinyUSB_Library: https://forums.adafruit.com/viewtopic.php?f=25&t=179789 :
 * --> "There is a major changes in TinyUSB lib from v1.0.0 that require core changes by xiao bsp."
 * 
 */

#include <uClock.h>

#include <Adafruit_TinyUSB.h>
Adafruit_USBD_MIDI usb_midi;

#include <MIDI.h>
using namespace midi;

MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, usbMIDI);

uint8_t bpm_blink_timer = 1;
void handle_bpm_led(uint32_t * tick)
{
  // BPM led indicator
  if ( !(*tick % (96)) || (*tick == 1) ) {  // first compass step will flash longer
    bpm_blink_timer = 8;
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(*tick % (24)) ) {   // each quarter led on
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(*tick % bpm_blink_timer) ) { // get led off
    digitalWrite(LED_BUILTIN, LOW);
    bpm_blink_timer = 1;
  }
}

// Internal clock handlers
void ClockOut96PPQN(uint32_t * tick) {
  // Send MIDI_CLOCK to external gears
  usbMIDI.sendRealTime(Clock);
  handle_bpm_led(tick);
}
[... 60 lines cut ...]
/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"2022-04-27-uClock-Xiao-sketch_apr27a.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10819 -DARDUINO_SEEED_XIAO_M0 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -D__SAMD21__ -D__SAMD21G18A__ -DARM_MATH_CM0PLUS -DSEEED_XIAO_M0 -DUSB_VID=0x2886 -DUSB_PID=0x802F -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Seeed\"" "-DUSB_PRODUCT=\"Seeed XIAO M0\"" -DUSE_TINYUSB -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -DARDUINO_SAMD_ZERO -D__SAMD21__ -D__SAMD21G18A__ -DARM_MATH_CM0PLUS -DSEEED_XIAO_M0 -DUSB_VID=0x2886 -DUSB_PID=0x802F -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Seeed\"" "-DUSB_PRODUCT=\"Seeed XIAO M0\"" -DUSE_TINYUSB -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0/CMSIS/DSP/Include/ -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS-Atmel/1.2.1/CMSIS-Atmel/CMSIS/Device/ATMEL/ -I/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/cores/arduino -I/Volumes/HDD/wetmath [... cut ...]
Using library Adafruit_TinyUSB_Library at version 0.10.5 in folder: /Volumes/HDD/wetmath/Documents/Arduino/libraries/Adafruit_TinyUSB_Library 
Using library MIDI_Library at version 5.0.2 in folder: /Volumes/HDD/wetmath/Documents/Arduino/libraries/MIDI_Library 
Using library TimerTCC0 in folder: /Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/libraries/TimerTCC0 (legacy)
exit status 1
invalid conversion from 'void (*)(uint32_t*) {aka void (*)(long unsigned int*)}' to 'void (*)(volatile uint32_t*) {aka void (*)(volatile long unsigned int*)}' [-fpermissive]
```

mightycoco, 2022-04-28:
You might have to add a volatile modifier to the method XiaoUsbMasterMidiClock.cpp
void ClockOut96PPQN(volatile uint32_t * tick) {... 

That seems to be a quirk in the SAMD implementation that volatile XY isn't castable as XY and the other way around. Tha's why I have this ugly block #if #else #endif block in uClock.h specifying different method initializers.

m-r-m-s, 2022-04-28:
> You might have to add a volatile modifier to the method XiaoUsbMasterMidiClock.cpp
> void ClockOut96PPQN(volatile uint32_t * tick) {...

Thanks! I tried changing to volatile and it compiles with the following modified sketch ("simple MIDI Sync Box on Teensy boards and USB Midi setup") provided by @midilab (see below).

However the XIAO is _not_ recognized as a USB MIDI device in MIDI monitor or in the MIDI set up on my Mac. 

> That seems to be a quirk in the SAMD implementation that volatile XY isn't castable as XY and the other way around. That's why I have this ugly block #if #else #endif block in uClock.h specifying different method initializers.

I am admittedly out of my depth here. Perhaps we have enough info now that @midilab needs? Specifically the MACRO you used (`SEEED_XIAO_M0`) for the for Xiao (which is also referenced [here](https://forum.seeedstudio.com/t/macro-for-distigushing-xiao-akin-to-arduino-arch-avr/262142/2)).

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>
using namespace midi;
#include <uClock.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;
 
// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// The callback function wich will be called by Clock each Pulse of 96PPQN clock resolution.
void ClockOut96PPQN(volatile uint32_t * tick) {
  // Send MIDI_CLOCK to external gears
  MIDI.sendRealTime(Clock);
}

// The callback function wich will be called when clock starts by using Clock.start() method.
void onClockStart() {
  MIDI.sendRealTime(Start);
}

// The callback function wich will be called when clock stops by using Clock.stop() method.
void onClockStop() {
  MIDI.sendRealTime(Stop);
}

void setup() {

    // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);
  
  // Inits the clock
  uClock.init();
  // Set the callback function for the clock output to send MIDI Sync message.
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // Set the callback function for MIDI Start and Stop messages.
  uClock.setOnClockStartOutput(onClockStart);  
  uClock.setOnClockStopOutput(onClockStop);
  // Set the clock BPM to 126 BPM
  uClock.setTempo(126);
  // Starts the clock, tick-tac-tick-tac...
  uClock.start();

  Serial.begin(115200);
 
  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);
}

// Do it whatever to interface with Clock.stop(), Clock.start(), Clock.setTempo() and integrate your environment...
void loop() {
  
}
```

Alternatively, when I use the XIAO MIDI example from [here](https://wiki.seeedstudio.com/Wio-Terminal-USBCLIENT-MIDI/#complete-code) it compiles and is recognized as a USB MIDI device on the Serial Monitor and in MIDI monitor:

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>
using namespace midi;

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;
 
// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
 
// Variable that holds the current position in the sequence.
uint32_t position = 0;
 
// Store example melody as an array of note values
byte note_sequence[] = {
  74,78,81,86,90,93,98,102,57,61,66,69,73,78,81,85,88,92,97,100,97,92,88,85,81,78,
  74,69,66,62,57,62,66,69,74,78,81,86,90,93,97,102,97,93,90,85,81,78,73,68,64,61,
  56,61,64,68,74,78,81,86,90,93,98,102
};
 
void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
 
  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);
 
  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);
 
  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);
 
  Serial.begin(115200);
 
  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);
}
 
void loop()
{
  static uint32_t start_ms = 0;
  if ( millis() - start_ms > 266 )
  {
    start_ms += 266;
 
    // Setup variables for the current and previous
    // positions in the note sequence.
    int previous = position - 1;
 
    // If we currently are at position 0, set the
    // previous position to the last note in the sequence.
    if (previous < 0) {
      previous = sizeof(note_sequence) - 1;
    }
 
    // Send Note On for current position at full velocity (127) on channel 1.
    MIDI.sendNoteOn(note_sequence[position], 127, 1);
 
    // Send Note Off for previous note.
    MIDI.sendNoteOff(note_sequence[previous], 0, 1);
 
    // Increment position
    position++;
 
    // If we are at the end of the sequence, start over.
    if (position >= sizeof(note_sequence)) {
      position = 0;
    }
  }
  // read any new incoming MIDI messages
  MIDI.read();  
}
 
void handleNoteOn(byte channel, byte pitch, byte velocity)
{
  // Log when a note is pressed.
  Serial.printf("Note on: channel = %d, pitch = %d, velocity - %d", channel, pitch, velocity);
  Serial.println();
}
 
void handleNoteOff(byte channel, byte pitch, byte velocity)
{
  // Log when a note is released.
  Serial.printf("Note off: channel = %d, pitch = %d, velocity - %d", channel, pitch, velocity);
  Serial.println();
}
```

midilab, 2022-05-02:
thanks for all the info, i was wondering if that could be a best option to code AVR and ARM flags instead of a lot of different platforms over there.
For what i see on @mightycoco support fod xiao they use a different call for timer setup... maybe we can go for AVR and ARM generic and those who need special handling like XIAO we can add a XIAO flag after ARM flag code block.
I will take a look soon and get back with the idea at some branch.

midilab, 2022-05-03:
ok so here it is @m-r-m-s , i didn't have a time so far to test... so let me know!

https://github.com/midilab/uClock/pull/9

So for the volatile warning i've fix it too, i forgot to take care and review those variables contexts after rewrite 1.0 engine. 

There were no need for the variable to be volatile now, so passing by value now instead of reference too.

Its a breakchange on uClock API... so i changed the examples too... if everything is ok from your tests i will release this xiao branch support.

thank you @mightycoco for the research and code for xiao support.

midilab, 2022-05-03:
> Thanks! I tried changing to volatile and it compiles with the following modified sketch ("simple MIDI Sync Box on Teensy boards and USB Midi setup") provided by @midilab (see below).
>
> However the [...]

Try to initialize the uClock after usb is ready:

while( !USBDevice.mounted() ) delay(1);
// ... uClock setup goes bellow...

m-r-m-s, 2022-05-03:
> ok so here it is @m-r-m-s , i didn't have a time so far to test... so let me know!
>
> #9
>
> So for the volatile warning i've fix it too, i forgot to take care and review those variables contexts after [...]

Thank you !!! Ok, I did a quick test with the USB MIDI Sync Box Example and I receive a compile error:
```
Arduino: 1.8.19 (Mac OS X), TD: 1.56, Board: "Seeeduino XIAO, TinyUSB, Off"

WARNING: library uClock-xiao-support claims to run on avr, arm architecture(s) and may be incompatible with your current board which runs on samd architecture(s).
/Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/2022-05-03-xiao-uclock-usb-midi-syncbox-example/2022-05-03-xiao-uclock-usb-midi-syncbox-example.ino: In function 'void ClockOut96PPQN(volatile uint32_t*)':
/Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/2022-05-03-xiao-uclock-usb-midi-syncbox-example/2022-05-03-xiao-uclock-usb-midi-syncbox-example.ino:36:41: warning: unused parameter 'tick' [-Wunused-parameter]
 void ClockOut96PPQN(volatile uint32_t * tick) {
                                         ^~~~
/Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/2022-05-03-xiao-uclock-usb-midi-syncbox-example/2022-05-03-xiao-uclock-usb-midi-syncbox-example.ino: In function 'void setup()':
2022-05-03-xiao-uclock-usb-midi-syncbox-example:65:45: error: invalid conversion from 'void (*)(volatile uint32_t*) {aka void (*)(volatile long unsigned int*)}' to 'void (*)(uint32_t) {aka void (*)(long unsigned int)}' [-fpermissive]
   uClock.setClock96PPQNOutput(ClockOut96PPQN);
                                             ^
In file included from /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/2022-05-03-xiao-uclock-usb-midi-syncbox-example/2022-05-03-xiao-uclock-usb-midi-syncbox-example.ino:13:0:
/Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/libraries/uClock-xiao-support/src/uClock.h:114:8: note:   initializing argument 1 of 'void umodular::clock::uClockClass::setClock96PPQNOutput(void (*)(uint32_t))'
   void setClock96PPQNOutput(void (*callback)(uint32_t tick)) {
        ^~~~~~~~~~~~~~~~~~~~
exit status 1
invalid conversion from 'void (*)(volatile uint32_t*) {aka void (*)(volatile long unsigned int*)}' to 'void (*)(uint32_t) {aka void (*)(long unsigned int)}' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
Here is the modified example:

```
#include <Arduino.h>

#include <Adafruit_TinyUSB.h>
// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

#include <MIDI.h>
using namespace midi;
// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

#include <uClock.h>

/* USB MIDI Sync Box
 *  
 * This example demonstrates how to change the USB MIDI 
 * device name on Teensy LC and 3.x.  When creating more
 * that one MIDI device, custom names are much easier to
 * use when selecting each device in MIDI software on
 * your PC or Mac.  The custom name is in the "name.c" tab.
 *
 * Windows and Macintosh systems often cache USB info.
 * After changing the name, you may need to test on a
 * different computer to observe the new name, or take
 * steps to get your operating system to "forget" the
 * cached info.  (TODO: wanted... can anyone contribute
 * instructions for these systems)
 * 
 * You must select MIDI from the "Tools > USB Type" menu
 * 
 * This example code is in the public domain.
 */

// The callback function wich will be called by Clock each Pulse of 96PPQN clock resolution.
void ClockOut96PPQN(volatile uint32_t * tick) {
  // Send MIDI_CLOCK to external gears
  MIDI.sendRealTime(Clock);
}

// The callback function wich will be called when clock starts by using Clock.start() method.
void onClockStart() {
  MIDI.sendRealTime(Start);
}

// The callback function wich will be called when clock stops by using Clock.stop() method.
void onClockStop() {
  MIDI.sendRealTime(Stop);
}

void setup() {

  Serial.begin(115200);
 
  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);

    // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);
  
  // Inits the clock
  uClock.init();
  // Set the callback function for the clock output to send MIDI Sync message.
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // Set the callback function for MIDI Start and Stop messages.
  uClock.setOnClockStartOutput(onClockStart);  
  uClock.setOnClockStopOutput(onClockStop);
  // Set the clock BPM to 126 BPM
  uClock.setTempo(126);
  // Starts the clock, tick-tac-tick-tac...
  uClock.start();
  
}

// Do it whatever to interface with Clock.stop(), Clock.start(), Clock.setTempo() and integrate your environment...
void loop() {
  
}
```

midilab, 2022-05-03:
The new version change the clock callbacks signatures, no more volatile or reference value so please change:

```
void ClockOut96PPQN(volatile uint32_t * tick) {
uint32_t value = *tick;
...
}

per
void ClockOut96PPQN(uint32_t tick) {
uint32_t value = tick;
...
}
```

m-r-m-s, 2022-05-03:
> The new version change the clock callbacks signatures, no more volatile or reference value so please change:
> 
> ```
> void ClockOut96PPQN(volatile uint32_t * tick) {
> uint32_t value = *tick;
> ...
> }
> 
> per
> void ClockOut96PPQN(uint32_t tick) {
> uint32_t value = tick;
> ...
> }
> ```

Ah ooopsie! I missed removing `volatile` . However, there is still a compile error:

```
Arduino: 1.8.19 (Mac OS X), TD: 1.56, Board: "Seeeduino XIAO, TinyUSB, Off"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Volumes/HDD/wetmath/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Volumes/HDD/wetmath/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -vid-pid=2886_802F -ide-version=10819 -build-path /var/folders/m4/cfg6cv292zq5pvk0lvpbqcv80000gn/T/arduino_build_329848 -warnings=all -build-cache /var/folders/m4/cfg6cv292zq5pvk0lvpbqcv80000gn/T/arduino_cache_737734 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.CMSIS-Atmel.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.arduinoOTA.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.bossac.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.CMSIS.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/bossac/1.7.0-arduino3 -verbose /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/2022-05-03-xiao-uclock-usb-midi-syncbox-example/2022-05-03-xiao-uclock-usb-midi-syncbox-example.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Volumes/HDD/wetmath/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Volumes/HDD/wetmath/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/libraries -fqbn=Seeeduino:samd:seeed_XIAO_m0:usbstack=tinyusb,debug=off -vid-pid=2886_802F -ide-version=10819 -build-path /var/folders/m4/cfg6cv292zq5pvk0lvpbqcv80000gn/T/arduino_build_329848 -warnings=all -build-cache /var/folders/m4/cfg6cv292zq5pvk0lvpbqcv80000gn/T/arduino_cache_737734 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.CMSIS-Atmel.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.arduinoOTA.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.bossac.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.CMSIS.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/tools/CMSIS/5.7.0 -prefs=runtime.tools.bossac-1.7.0-arduino3.path= [... cut ...]
Using library MIDI_Library at version 5.0.2 in folder: /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/libraries/MIDI_Library 
Using library uClock-xiao-support at version 1.0.0 in folder: /Volumes/HDD/wetmath/Documents/Arduino/00_Arduino_Dev/libraries/uClock-xiao-support 
Using library TimerTCC0 in folder: /Volumes/HDD/wetmath/Library/Arduino15/packages/Seeeduino/hardware/samd/1.8.2/libraries/TimerTCC0 (legacy)
exit status 1
invalid conversion from 'void (*)(uint32_t*) {aka void (*)(long unsigned int*)}' to 'void (*)(uint32_t) {aka void (*)(long unsigned int)}' [-fpermissive]

```

midilab, 2022-05-03:
you're still keeping the reference as a pointer.

it is:
 void ClockOut96PPQN(uint32_t tick) {
instead of:
 void ClockOut96PPQN(uint32_t * tick) {

see the *?

m-r-m-s, 2022-05-03:
> you're still keeping the reference as a pointer.
> 
> it is: void ClockOut96PPQN(uint32_t tick) { instead of: void ClockOut96PPQN(uint32_t * tick) {
> 
> see the *?

Oooof - too much or too little coffee, haha. I missed the pointer. Looks like you are making some updates to the XIAO-support branch so I can hold off testing for now. I haven't been able to get the XIAO to be recognized as a USB MIDI device yet. Just to make sure it wasn't my XIAO or the circuit I tested against this example and it works with no issues: 
```

/*
                   _ _   _
                  | | | (_)
   _ __ ___  _   _| | |_ _
  | '_ ` _ \| | | | | __| |
  | | | | | | |_| | | |_| |
  |_| |_| |_|\__,_|_|\__|_|

  https://github.com/pangrus/multi/tree/main/multi_software/midi_converter
  midi converter 
  USB MIDI <---> MIDI DIN

  CC BY-NC-SA
  pangrus 2021
*/

#include <Adafruit_TinyUSB.h> // XIAO only supports version 0.10.5 of Adafruit TinyUSB Library for Arduino 
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI_USB);
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI_DIN);

void setup() {
  SerialUSB.begin(115200);
  MIDI_USB.begin(MIDI_CHANNEL_OMNI);
  MIDI_USB.setHandleStart(UsbHandleStart);
  MIDI_USB.setHandleStop(UsbHandleStop);
  MIDI_USB.setHandleNoteOn(UsbHandleNoteOn);
  MIDI_USB.setHandleNoteOff(UsbHandleNoteOff);
  MIDI_USB.setHandleClock(UsbHandleClock);
  MIDI_USB.setHandleControlChange(UsbHandleCC);
  MIDI_DIN.begin(MIDI_CHANNEL_OMNI);
  MIDI_DIN.setHandleStart(DinHandleStart);
  MIDI_DIN.setHandleStop(DinHandleStop);
  MIDI_DIN.setHandleNoteOn(DinHandleNoteOn);
  MIDI_DIN.setHandleNoteOff(DinHandleNoteOff);
  MIDI_DIN.setHandleClock(DinHandleClock);
  MIDI_DIN.setHandleControlChange(DinHandleCC);
  MIDI_DIN.turnThruOff();
}

void loop() {
  MIDI_USB.read();
  MIDI_DIN.read();
}

void UsbHandleStart() {
  MIDI_DIN.sendStart();
  SerialUSB.println("USB --> DIN MIDI Start");
}

void UsbHandleStop() {
  MIDI_DIN.sendStop();
  SerialUSB.println("USB --> DIN MIDI Stop");
}

void UsbHandleClock() {
  MIDI_DIN.sendClock();
}

void UsbHandleNoteOn(byte channel, byte note, byte velocity) {
  MIDI_DIN.sendNoteOn(note, velocity, channel);
  SerialUSB.print("USB --> DIN Note on: ");
  SerialUSB.print(note);
  SerialUSB.print(" Channel: ");
  SerialUSB.print(channel);
  SerialUSB.print(" Velocity: ");
  SerialUSB.println(velocity);
}

void UsbHandleNoteOff(byte channel, byte note, byte velocity) {
  MIDI_DIN.sendNoteOff(note, velocity, channel);
}

void UsbHandleCC(byte channel, byte control, byte value) {
  MIDI_DIN.sendControlChange(control, value, channel);
  SerialUSB.print("USB --> DIN Control Change: ");
  SerialUSB.print(control);
  SerialUSB.print(" Channel: ");
  SerialUSB.print(channel);
  SerialUSB.print(" Value: ");
  SerialUSB.println(value);
}

void DinHandleStart() {
  MIDI_USB.sendStart();
  SerialUSB.println("DIN --> USB MIDI Start");
}

void DinHandleStop() {
  MIDI_USB.sendStop();
  SerialUSB.println("DIN --> USB MIDI Stop");
}

void DinHandleClock() {
  MIDI_USB.sendClock();
}

void DinHandleNoteOn(byte channel, byte note, byte velocity) {
  MIDI_USB.sendNoteOn(note, velocity, channel);
  SerialUSB.print("DIN --> USB Note on: ");
  SerialUSB.print(note);
  SerialUSB.print(" Channel: ");
  SerialUSB.print(channel);
  SerialUSB.print(" Velocity: ");
  SerialUSB.println(velocity);
}

void DinHandleNoteOff(byte channel, byte note, byte velocity) {
  MIDI_USB.sendNoteOff(note, velocity, channel);
  SerialUSB.print("DIN --> USB Note off: ");
  SerialUSB.print(note);
  SerialUSB.print(" Channel: ");
  SerialUSB.print(channel);
  SerialUSB.print(" Velocity: ");
  SerialUSB.println(velocity);
}

void DinHandleCC(byte channel, byte control, byte value) {
  MIDI_USB.sendControlChange(control, value, channel);
  SerialUSB.print("DIN --> USB Control Change: ");
  SerialUSB.print(control);
  SerialUSB.print(" Channel: ");
  SerialUSB.print(channel);
  SerialUSB.print(" Value: ");
  SerialUSB.println(value);
}
```

midilab, 2022-05-03:
just two minor updates, one to add samd as architecture at libraries.properties to avoid a warning at compile time for xiao, and other for README and docs in general, i need to test at my bords too to check if everything is ok.

Try to debug specific parts of the code to check if the initialization of usb is ok, print some debug lines after parts of setup() call, specially that while waiting for the USB to be ready, maybe its looping there forever...

m-r-m-s, 2022-05-03:
Sounds good - thank you. I'll do my best to try and debug and report back. Thanks!

midilab, 2022-05-03:
if you can test with the latest commit on the branch it will be great!

m-r-m-s, 2022-05-03:
I was not able to print to the serial monitor unfortunately with the latest commit. It compiles and uploads BUT the Seeeduino Xiao USB port disappears after upload which indicates a failure even though it says it compiles and uploads.  I have to then reset to bootloader mode for the port to reappear and then upload the sketch. This happens with or without `while( !USBDevice.mounted() ) delay(1);` in the setup. This does not happen in this sketch: [https://github.com/pangrus/multi/tree/main/multi_software/midi_converter](https://github.com/pangrus/multi/tree/main/multi_software/midi_converter)

Do you have access to a XIAO or just Teensy and Arduino? I am happy to keep testing for you if you do not have a XIAO. I'll need some guidance on how to debug however. Thanks!

Here is the sketch I tried:

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h> // NOTE: only use version 0.10.5! See: https://wiki.seeedstudio.com/Seeeduino-XIAO-TinyUSB/
#include <MIDI.h>
using namespace midi; // https://forum.arduino.cc/t/midi-library-and-midi-sendrealtime-function/223760/11

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, usbMIDI);

/* USB MIDI Sync Slave Box
 * https://github.com/midilab/uClock/blob/xiao-support/examples/TeensyUsbSlaveMidiClock/TeensyUsbSlaveMidiClock.ino
 * This example demonstrates how to change the USB MIDI 
 * device name on Teensy LC and 3.x.  When creating more
 * that one MIDI device, custom names are much easier to
 * use when selecting each device in MIDI software on
 * your PC or Mac.  The custom name is in the "name.c" tab.
 *
 * Windows and Macintosh systems often cache USB info.
 * After changing the name, you may need to test on a
 * different computer to observe the new name, or take
 * steps to get your operating system to "forget" the
 * cached info.  (TODO: wanted... can anyone contribute
 * instructions for these systems)
 * 
 * You must select MIDI from the "Tools > USB Type" menu
 * 
 * This example code is in the public domain.
 */

#include <uClock.h>

uint8_t bpm_blink_timer = 1;
void handle_bpm_led(uint32_t tick)
{
  // BPM led indicator
  if ( !(tick % (96)) || (tick == 1) ) {  // first compass step will flash longer
    bpm_blink_timer = 8;
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % (24)) ) {   // each quarter led on
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % bpm_blink_timer) ) { // get led off
    digitalWrite(LED_BUILTIN, LOW);
    bpm_blink_timer = 1;
  }
}

// Internal clock handlers
void ClockOut96PPQN(uint32_t tick) {
  // Send MIDI_CLOCK to external gears on other port?
  //usbMIDI.sendRealTime(usbMIDI.Clock);
  //usbMIDI.sendRealTime(Clock);
  usbMIDI.sendClock();
  handle_bpm_led(tick);
}

void onClockStart() {
  //usbMIDI.sendRealTime(usbMIDI.Start);
  //usbMIDI.sendRealTime(Start);
  usbMIDI.sendStart();
}

void onClockStop() {
  //usbMIDI.sendRealTime(usbMIDI.Stop);
  //usbMIDI.sendRealTime(Stop);
  usbMIDI.sendStop();
}

// External clock handlers
void onExternalClock()
{
  uClock.clockMe();
}

void onExternalStart()
{
  uClock.start();
}

void onExternalStop()
{
  uClock.stop();
}

void setup() {
  // A led to count bpms
  pinMode(LED_BUILTIN, OUTPUT);
  
  Serial.print(USBDevice.mounted());
  Serial.print("USBDevice.mounted(): ");
  Serial.println();
  
  // Setup realtime midi event handlers
  usbMIDI.setHandleClock(onExternalClock);
  usbMIDI.setHandleStart(onExternalStart);
  usbMIDI.setHandleStop(onExternalStop);

  // Setup our clock system
  // Inits the clock
  uClock.init();
  // Set the callback function for the clock output to send MIDI Sync message.
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // Set the callback function for MIDI Start and Stop messages.
  uClock.setOnClockStartOutput(onClockStart);  
  uClock.setOnClockStopOutput(onClockStop);
  // set to external sync mode
  uClock.setMode(1);
}

void loop() {
  // Grab all midi data as fast as we can!
  while (usbMIDI.read()) {}
 
}
```

midilab, 2022-05-04:
first thing i note here is that you code is different from the one that works, on usbMidi setup()
the one that works inits usbMidi before any other call.
MIDI_USB.begin(MIDI_CHANNEL_OMNI);
please comment all uClock related code first on your code, and try to make it work with usbMidi only.
if you get it work, uncomment uclock code and see what happens.

mightycoco, 2022-05-04:
I had to change a few things, to get it work. I'm not sure if this is specific to my implementation, as I'm also attaching other Hardware Interrupts to some methods unrelated to the uClock implementation.

Instead of makeing a reference to TimerTCC0 via
TimerTCC0 _uclockTimer;
I'm directly accessing TimerTcc0 defined in TimerTCC0.cpp like
TimerTcc0.initialize(init_clock);

My XIAO did hang up, when using 
_uclockTimer.attachInterrupt(uclockISR);
But works when using
TimerTcc0.attachInterrupt(uclockISR);

See this example to make things clear: https://github.com/mightycoco/uClock/commit/5edd58fcdc157de5990ad1b442ee57e2c885316a

midilab, 2022-05-04:
uhmmm interesting, thanks for the implementation @mightycoco, i've just update the branch with your implementation changes.
@m-r-m-s , maybe that helps!

midilab, 2022-05-04:
@mightycoco also feel free to use the pull request i've created for that xiao support feature for your contributions.

m-r-m-s, 2022-05-04:
I have tested with @mightycoco  implementation with the [uClock example above](https://github.com/midilab/uClock/issues/6#issuecomment-1116703998) and no results: the USB MIDI device is still not recognized.

midilab, 2022-05-04:
also try to comment out the 
// handle_bpm_led(tick);
i remember to have problems with it on a specific MCU...

midilab, 2022-05-04:
another thing to try is to comment our all usbMidi calls inside uClock callbacks, and print something instead... if something prints out its a good sign...

m-r-m-s, 2022-05-04:
> maybe this usb library are making use of timer0 or another dependency on your project?

I checked the [source code](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases/tag/0.10.5) for this USB Library and I couldn't find any references to timer0 or TimerTCC0.

> also try to comment out the
> // handle_bpm_led(tick);
> i remember to have problems with it on a specific MCU...

> another thing to try is to comment our all usbMidi calls inside uClock callbacks, and print something instead... if something prints out its a good sign...

I tried both of these things and the USB MIDI device is still not recognized unfortunately.

midilab, 2022-05-04:
daaam, thats weird...
can you run a uClock test on XIAO without usb library? jsut to make sure it works without the usb library? then its a good start point to continue debuging...

m-r-m-s, 2022-05-04:
Ok, sure! Do you mean just run a uClock test on XIAO as MIDI DIN correct?

midilab, 2022-05-04:
yes! that will be great! so we know that XIAO support is working at most.
so i can merge those changes and we keep trying here to get a solution from another branch

m-r-m-s, 2022-05-04:
Ok, unfortunately, I have had no success on the xio-support branch for uclock with MIDI DIN. I double checked my MIDI DIN circuit on the XIAO breadboard with a basic test with only the MIDI library and it works with no issue. Here is the uClock sketch I tried:

```
/* USB MIDI Sync Slave Box
 *  https://raw.githubusercontent.com/midilab/uClock/xiao-support/examples/TeensyUsbSlaveMidiClock/TeensyUsbSlaveMidiClock.ino
 */
#include <Arduino.h>

#include <MIDI.h>

// Creates an instance of the MIDI library attached to a serial port.
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI);

#include <uClock.h>

uint8_t bpm_blink_timer = 1;
void handle_bpm_led(uint32_t tick)
{
  // BPM led indicator
  if ( !(tick % (96)) || (tick == 1) ) {  // first compass step will flash longer
    bpm_blink_timer = 8;
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % (24)) ) {   // each quarter led on
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % bpm_blink_timer) ) { // get led off
    digitalWrite(LED_BUILTIN, LOW);
    bpm_blink_timer = 1;
  }
}

// Internal clock handlers
void ClockOut96PPQN(uint32_t tick) {
  
//  handle_bpm_led(tick);
  Serial.print("CLOCK");
}

void onClockStart() {
  Serial.print("START");
}

void onClockStop() {
  Serial.print("STOP");
}

// External clock handlers
void onExternalClock()
{
  uClock.clockMe();
}

void onExternalStart()
{
  uClock.start();
}

void onExternalStop()
{
  uClock.stop();
}

void setup() {
  // A led to count bpms
  pinMode(LED_BUILTIN, OUTPUT);

  // We want to receive messages on all channels
  MIDI.begin(MIDI_CHANNEL_OMNI);
  
  // Setup realtime midi event handlers
  MIDI.setHandleClock(onExternalClock);
  MIDI.setHandleStart(onExternalStart);
  MIDI.setHandleStop(onExternalStop);

  // Setup our clock system
  // Inits the clock
  uClock.init();
  // Set the callback function for the clock output to send MIDI Sync message.
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // Set the callback function for MIDI Start and Stop messages.
  uClock.setOnClockStartOutput(onClockStart);  
  uClock.setOnClockStopOutput(onClockStop);
  // set to external sync mode
  uClock.setMode(1);
}

void loop() {
  // Grab all midi data as fast as we can!
//  while (MIDI.read()) {}
  MIDI.read();
}
```
For reference I tried the sketch (see it below) with a Teensy LC on uClock 0.10.5,  1.0, and the Xiao support branch 1.1 and they all work with no issue. One thing I am curious about is the `IntervalTimer` code in that sketch in the setup here:

```
uClock.setMode(uClock.EXTERNAL_CLOCK);
  // make use of 250us timer to handle midi input sync
  teensyTimer.begin(handleMidiInput, 250); 
  teensyTimer.priority(80);
```

and the fact that MIDI.read(); is not in the loop but here:

```
void handleMidiInput() {
  while (MIDI.read()) {
  }
}
```

My novice question: Even though the XIAO support branch accesses timer0 now directly,  don't we need some type of ISR like the intervalTimer? Or is that already happening somewhere else in uClock?

Here's the working uClock sketch on Teensy LC (v.0 & 0.10.5 & xiao-support branch 1.1):

```
// THIS WORKS WITH UCLOCK VERSION 1.0 & 0.10.5 & xiao-support branch 1.1 ON TEENSY LC

#include <Arduino.h>
#include <MIDI.h>

// Creates an instance of the MIDI library attached to a serial port.
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI);

#include <uClock.h>
IntervalTimer teensyTimer;

// Internal clock handlers
void ClockOut96PPQN(uint32_t tick) {
  
}

void onClockStart() {

}

void onClockStop() {

}

// External clock handlers
void onExternalClock() {
  uClock.clockMe();
}

void onExternalStart() {
  uClock.start();
}

void onExternalStop() {
  uClock.stop(); 
}

void setup() {

  // MIDI setup
  
  // We want to receive messages on all channels
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // MIDI library clock handlers for MIDI Sync Start and Stop
  MIDI.setHandleClock(onExternalClock);
  MIDI.setHandleStart(onExternalStart);
  MIDI.setHandleStop(onExternalStop);

  // uClock Setup
  // Setup our clock system
  uClock.init();
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // For MIDI Sync Start and Stop
  uClock.setOnClockStartOutput(onClockStart);
  uClock.setOnClockStopOutput(onClockStop);
  
  uClock.setMode(uClock.EXTERNAL_CLOCK); // uClock.setMode(1) also works
  // make use of 250us timer to handle midi input sync
  teensyTimer.begin(handleMidiInput, 250); 
  teensyTimer.priority(80);

}

void handleMidiInput() {
  while (MIDI.read()) {
  }
}

void loop() {
  
}
```

midilab, 2022-05-08:
Just to confirm, does the uClock without MIDI library works on your XIAO tests(xiao-support branch)? uClock solo(it can be a master sync test with no slave mode).

The IntervalTimer is used to read MIDI input at the speed of 250us.That way we have a predictable way of reading midi input in realtime to garantee the best sync slave timming as possible. 

Using loop() for midi read is ok if MIDI.read is the only call at. the more code you put on loop() the more unpredictable it will be the MIDI.read inside it and so the slave timming sync.

midilab, 2022-05-09:
I've created a xiao example that will be good start point to debug the problem. can you try it and check if the usb midi device will work along with uclock?
https://github.com/midilab/uClock/blob/xiao-support/examples/XiaoUsbMasterMidiClock/XiaoUsbMasterMidiClock.ino

m-r-m-s, 2022-05-10:
> The IntervalTimer is used to read MIDI input at the speed of 250us.That way we have a predictable way of reading midi input in realtime to garantee the best sync slave timming as possible.
>
> Using [...]

Excellent - thank you for explaining these to me. Very helpful! 

> Just to confirm, does the uClock without MIDI library works on your XIAO tests(xiao-support branch)? uClock solo(it can be a master sync test with no slave mode).

> I've created a xiao example that will be good start point to debug the problem. can you try it and check if the usb midi device will work along with uclock?
> https://github.com/midilab/uClock/blob/xiao-support/examples/XiaoUsbMasterMidiClock/XiaoUsbMasterMidiClock.ino

I believe I tried previously but I am going to try the xiao example you created to help debug. I'll report back here as soon as I can and let you know. Thanks!

m-r-m-s, 2022-05-13:
> I've created a xiao example that will be good start point to debug the problem. can you try it and check if the usb midi device will work along with uclock? https://github.com/midilab/uClock/blob/xiao-support/examples/XiaoUsbMasterMidiClock/XiaoUsbMasterMidiClock.ino

I've tried the example and the Xiao still does not show up as a USB MIDI device in MIDI monitor or in MIDI studio on my Mac. If I comment out all of the uClock code in the example, it is then recognized as a USB MIDI device. If you can let me know what info you need that would be helpful to further debug, please let me know and I can provide that. Thank you!

midilab, 2022-05-17:
Great! i couldn't find the docs for their timers unfortanly, but seeing some examples there is another timer called Tc3... i code uclock using it we can test with a different timer, maybe tcc0 is already in use for another part of arduino library just like timer0 on avrs...

test the new commit at xiao-support and let me know please? thank you!

mightycoco, 2022-05-18:
Side note: Tc3 has a resolution of 16 bit instead of Tcc0 with 24 bit. But i guess for a midi implementation 16 bit should be more than enough.

I haven't found any issues with your previous implementations and it's runnign straight and steady with both external and internal clock signals.

I'm not using any MIDI implementations, though - but straight analog clock-in signals triggered by a HW interrupt on GPIO 3.

midilab, 2022-05-18:
great to know that works @mightycoco, so im going to release the code with Tcc0 and leave a commented for Tc3 in case anyone would like to use along with other library that can possible make use of Tcc0.

Its ok 16its for clock on midi at 96ppqn, but its better for sure a 24bits!

@m-r-m-s we can keep trying thru here a solution for the MIDI usb along with uclock.

m-r-m-s, 2022-05-18:
@midilab thanks for doing this and pushing the new version! I am going to try and test today and I will report back.

> I was searching for xiao arduino support on timers programming but i couldn't anything unfortunaly, but seeing some examples there is another timer called Tc3... so i code uclock using him, so we can test with a different timer, maybe tcc0 is already in use for another part of arduino library just like timer0 on avrs...

In case you haven't come across this page, this is a pretty extensive overview of the Xiao including timers. The repo has lots of examples alongside this overview: 
[https://sigmdel.ca/michel/ha/xiao/seeeduino_xiao_01_en.html#timers](https://sigmdel.ca/michel/ha/xiao/seeeduino_xiao_01_en.html#timers)

https://github.com/sigmdel/xiao_m0_sketches

m-r-m-s, 2022-05-18:
@midilab I tried the [XiaoUsbMasterMidiClock](https://github.com/midilab/uClock/tree/master/examples/XiaoUsbMasterMidiClock) with uClock  v1.1.0 with tcc0 lines commented and tc3  lines uncommented and it is the same behavior as before where the USB MIDI device is still not recognized. 

Is it possible this issue has something to do with some sort of conflict with the USB stack and not the timer? 

The Xiao requires this specific version of TinyUSB for reference: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases/tag/0.10.5

To clarify: my own specific use case for uClock does not _require_ it to have USB MIDI. I am using a modified version of the https://github.com/midilab/uClock/tree/master/examples/TeensyUsbSlaveMidiClockMonitor example for DIN MIDI.

My thinking was to get the USB version of uClock working first and then modify for DIN...

midilab, 2022-05-18:
the seedstudio arduino implementation could be using some timer like tcc0 or tc3 for any internal function(like millis() or delay()) or USB implmementaion is my best guess. That's why its good to have a option to switch to a free timer in case...

Its hard to test without the XIAO in hands. But i always test things minizing code, the minimun we know that works so far is uClock on Xiao, without usb midi...

Try to add usb midi or DIN in parts from that point.

Try to add things without call uClock.init() for example can gives a nice clue about the problem.

pangrus, 2022-06-01:
Hi @midilab, @mightycoco, @eefh 
I have a [project based on Xiao](https://github.com/pangrus/multi) and I get in touch with @m-r-m-s so I'm here to help in the debugging process for the uclock library.
I started with the bare minimum.
```
#include <uClock.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
I do not get any error compiling/loading but, as @m-r-m-s noted:

> the Seeeduino Xiao USB port disappears after upload which indicates a failure
I also tried to use Timer TC3 instead of TC0 uncommenting the lines that @midilab has already prepared but nothing changes.

I also have written a sketch that uses Timer TC3, USB MIDI and DIN MIDI succesfully, that indicates there's not incompatibility between the libraries involved.
`
/*
  midi test
  CC BY-NC-SA
  pangrus 2022
*/

#include <MIDI.h>
#include <Adafruit_TinyUSB.h>
#include <TimerTC3.h>
#define USE_TIMER_TC3   // use TimerTc3
#define TIMER_TIME 125  // 1/8000 Hz = 125 micro seconds
#define TIMER TimerTc3
#define DAC_PIN 0       // pin of the XIAO DAC
Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI_USB);
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI_DIN);
void setup () {
  TIMER.initialize(TIMER_TIME);
  TIMER.attachInterrupt(GenerateClock);
  Serial.begin(9600);
  SerialUSB.begin(115200);
  MIDI_USB.begin(MIDI_CHANNEL_OMNI);
  MIDI_DIN.begin(MIDI_CHANNEL_OMNI);
}
void loop () {
}
void GenerateClock() {
  clockTicker++;
  if (clockTicker >= 500) {
    clockTicker = 0;
    MIDI_DIN.sendClock();
  }
}
`
If I can help with further tests, let me know.

midilab, 2022-06-03:
hey @pangrus great project! 

Im getting really curious about xiao, nice price, nice features, i think i will need to order one for my self.

Thank you for the tests.

I've created a branch to try to solve this issue.

Can you try the same example you just test but using MIDI_USB.sendClock(); instead of MIDI_DIN.sendClock();?

Also i got some changes on the xiao example that can be used as a test too.
https://github.com/midilab/uClock/blob/xiao-usbmidi/examples/XiaoUsbMasterMidiClock/XiaoUsbMasterMidiClock.ino

pangrus, 2022-06-04:
I tested the new branch with the minimal sketch:
```
#include <uClock.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
No compiling errors, but after loading the XIAO is not recognised anymore.

midilab, 2022-06-06:
Strange, i couldn't figure out what could be the problem.

But i just order one for my self, so i can do some tests and debuging in here, i hope to get it work! its a real nice microcontroller for the price!

m-r-m-s, 2022-06-07:
Thanks for testing on your end @pangrus and look forward to when you receive the Xiao @midilab !

midilab, 2022-06-10:
Good news @m-r-m-s , i manage to get it work, actually was a realy simple problem, one line of code solve the issue, it was freezing when uClock tries to setup tempo using timer before initialize the timer.

So you can download the new release with this issue bugfixed:
https://github.com/midilab/uClock/releases/tag/v1.1.1

You can make use of Xiao example as start point:
https://github.com/midilab/uClock/blob/master/examples/XiaoUsbMasterMidiClock/XiaoUsbMasterMidiClock.ino

m-r-m-s, 2022-06-11:
That is great news @midilab ! Thank you. Tagging @pangrus here to also let them know the freezing issue has been resolved. I will try on my end here soon.

Does this mean that it works with both Timer `TCC0` and `TCC3` but with different bit resolutions?

pangrus, 2022-06-14:
Thanks @m-r-m-s for tagging me and thanks @midilab for your patience and effort to solve the issue. It works properly now.
Please note that on the XIAO it's necessary to write a 0 (LOW) to the pin to turn the LED_BUILTIN on.
Cheers!

midilab, 2022-06-14:
> 

@pangrus that explain why the builtin led was acting strange on xiao example i wrote! thanks for note that, i will commit a fix.

Thank you guys for testing out! 

Im closing this issue now.

closed 2022-06-14 by midilab