midilab/uClock #8: Multiple warnings when compiling uClock for Arduino

stevehunt100, 2022-05-01:
Even a simple sketch like:
#include <uClock.h> 

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

generates the following warnings...  Is there something I'm doing wrong?  Thanks

F:\Documents\Arduino\libraries\uClock-master\src\uClock.cpp: In member function 'void umodular::clock::uClockClass::handleTimerInt()':
F:\Documents\Arduino\libraries\uClock-master\src\uClock.cpp:358:39: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
   onClock96PPQNCallback(&internal_tick);
                                       ^
F:\Documents\Arduino\libraries\uClock-master\src\uClock.cpp:363:42: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
    onClock32PPQNCallback(&div32th_counter);
                                          ^
F:\Documents\Arduino\libraries\uClock-master\src\uClock.cpp:366:42: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
    onClock16PPQNCallback(&div16th_counter);
                                          ^
F:\Documents\Arduino\libraries\uClock-master\src\uClock.cpp:374:42: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
    onClock32PPQNCallback(&div32th_counter);
                                          ^

mightycoco, 2022-05-02:
Are you compiling for samd21 or any other cortex chip?

stevehunt100, 2022-05-02:
Sorry I should have said: yes it's for a Teensy 3.6 which uses a Cortex M4.  I'm using the standard Arduino compiler with Teensyduino extension.  Thanks.

midilab, 2022-05-02:
well technicly this code is wrong(uClock library), since the tick is a volatile and the call that use it are not volatile at function stack level. So the compiler is rigth warning you if you have -permisive flag set on your arduino environment. Nothing to worry at runtime, but it need a fix.
i will code a fix for that, maybe there is no need to make it volatile anyway, i will pull asap the changes.
thanks for report! mine arduino teensy doesn't get -fpermisive as default so i didn't catch this warning on my side.

stevehunt100, 2022-05-02:
Excellent - thanks Midilab :)  I don't understand this fully, but I thought that volatile should always be used on any variable that can be modified by an interrupt?

midilab, 2022-05-02:
volatile tells the compiler to avoid optmize code on a volatile variable handling.
the main usage for is to avoid non atomic operations over the same var when use it at interrupt and outside a interruption.

midilab, 2022-05-03:
This branch has a fix for the warnings...

https://github.com/midilab/uClock/pull/9

Soon it will be merged into master for a new release.

There is a breakchange here, now the tick is passed by value instead of by reference, so i changed the examples too with the correct new way of define you clock callbacks...
```

// before...
void ClockOut16PPQN(uint32_t * tick) {
...
uint32_t some_var = *tick;
...
}

// now
void ClockOut16PPQN(uint32_t tick) {
...
uint32_t some_var = tick;
...
}
```

stevehunt100, 2022-05-03:
Awesome thanks :)

midilab, 2022-05-18:
Hey @stevehunt100, the warnings messages were fixed at this new release:
https://github.com/midilab/uClock/releases/tag/v1.1.0

closed 2022-05-18 by midilab