midilab/uClock #10: uClock supported by Raspi pico ?

AlpesMachines, 2022-06-09:
Hi, 
i am already using you marvellous library to run a sequencer/arpeggiator on Arduino Mega and it is particularly reliable.
Is the uClock library compatible with the RP2040 used in Raspberry pico boards ? it 's a dual core ARM Cortex M0+ @133MHz 
Best regards

JV

midilab, 2022-06-10:
Hi @AlpesMachines , great device the matrix_ctrlr! do you have any video or sounds of generated arpeggios?

Are you talking about the Xiao RP2040? I've just wrote a support for the Xiao M0, but unfortunaly i dont have this other one with me to check how can it be ported.

Maybe i can try to get some info over the net about timers usage for RP2040 xiao arduino support and sent to you for tests.

AlpesMachines, 2022-06-11:
Hi @midilab 
thanks for your nice message ! the only [video](https://www.youtube.com/watch?v=HCbKbsIuvxo) i did has the Matrix Ctrlr in slave mode (external midi sync sent by Roland TR8)
I am not talking about the specific Xiao RP2040 as the board has a very limited number of pins but this [one](https://forums.raspberrypi.com/viewtopic.php?t=326528) by WeAct which is a clone of the Raspi Pico.

I would be very happy and honoured to test your sketch 👍

midilab, 2022-06-11:
this oberheim sounds so good! congrats on the project, the interface is really amazing, how did you made the frontpanel?

So basiclly uClock is wrote for arduino platform only, if they come to support arduino(i didn't see any mention on the forum about it) then it will be easy to port.

Otherwise you can port the library to anyother platform since its basicly c++, but outsite arduino environment you need to figureout the timer usage and interrupt handlers of that platform to port it.

AlpesMachines, 2022-06-11:
thanks for your approval on my design 😃 
the front panel is a face plate in aluminium (but you could use PCB FR4 material) with a sticker on it (polyester V200) exactly like the Matrix6 and some car paint.
I know that uClock is based on arduino, especially TimerOne of AVR 8bits (that's why i am using it, as a replacement of the flexitimer2 lib), but has it is just a timer parameters definition, i would like it to be ported on other platform like STM32 or RP2040. Unfortunately i am not skilled enough within Timer to easily port that myself in just one evening 👅 

Hence my question as your development saves me lots of time ❤️

midilab, 2022-06-28:
Once they have some arduino port that could be viable to do.

closed 2022-06-28 by midilab

doctea, 2023-11-20:
Hi @midilab, thanks for this great library.  I recently managed to get it working with my Teensy 4.1 project where it has improved my sequencer's clock accuracy substantially compared to the naïve "track micros-since-last-tick" method.

Since I'm sharing a lot of the code between my Teensy project and my RP2040 projects, and even though the RP2040 project clock is much more accurate even using the naïve approach (I think because I can do all my GUI updates on the second core), I thought it might be good to get uClock working on the RP2040 as well.

I've had a little tinker, and tried to adapt the samd or esp32 platform headers to compile on the RP2040 using the earlephilhower rp2040 core, but not had any luck so far.  The earlephilhower core is built on top of freertos, so I'd hoped that the esp32 platform would be the least bother, but compile fails with eg "'timerBegin' was not declared in this scope".

Would porting uClock to rp2040 be simple at this stage?  Are there any hints you can give me on how to get this up and running?

Thanks!