midilab/uClock #12: Renaming the default branch from master

m-r-m-s, 2022-06-22:
Hi @midilab ,

Would you be willing to rename the default branch to `main` from `master`? The master-slave terminology has references to human chattel slavery. Github has already begun rolling out this change for the default to `main` for new repositories.

For reference: 

https://github.com/github/renaming and https://github.com/github/renaming#renaming-existing-branches

also:
https://cdm.link/2020/06/lets-dump-master-slave-terms/

I realize this would also apply to the uClock code base as well i.e. examples and other references so perhaps I can do a pull request or someone else in order to help out? Using the terms `main` and `secondary` seem like good replacement terminology.

Thanks for your consideration and again for all the great work!

midilab, 2022-08-10:
i read about some problems with commit history and forks related to change the main branch name, but i will take a time to carefull read about a way of doing without causing troubles.

m-r-m-s, 2022-08-11:
Ok, thanks so much!

midilab, 2022-10-27:
changed, in case anyone get any errors with local repo, just delete it and clone it again.

closed 2022-10-27 by midilab