midilab/uClock #13: Update Platformio library registry to latest uClock version?

m-r-m-s, 2022-08-09:
Hi @midilab ,
This is not urgent but if it is possible to update uClock to the latest version in the Platformio Library Registry, it would be greatly appreciated. It is currently at version `0.10.6` https://registry.platformio.org/libraries/midilab/uClock/versions
The registry should show midilab/uClock@^1.1.1` instead of `midilab/uClock@^0.10.6`

I am able to install the latest uClock version manually in the _platformio.ini_ file if I pull the repo directly (shown below) but others wanting the latest version of uClock may end up installing older version from the Platformio Registry. Thank you!
```
[env:teensylc]
platform = teensy
board = teensylc
framework = arduino
lib_deps =  https://github.com/midilab/uClock.git
```

midilab, 2022-08-10:
hey @m-r-m-s,

The best way of doing this is via pull request, send one with your changes after a last test and i will merge it.

m-r-m-s, 2022-08-11:
hey @midilab Sounds good! I am testing latest uClock version now and will report back. Thanks!

midilab, 2022-09-24:
in that case you can pull the changes to support from your side, as long as all pull is tested also from your side, because i dont have platformio to later test and dont want to support it by my self.