midilab/uClock #16: potential bug in TeensyUsbSlaveMidiClockMonitor example

m-r-m-s, 2022-08-11:
Hey @midilab I realized I didn't test uClock's latest version on the Teensy LC when we were testing the Xiao compatibility.

I went back and tested the following examples:
```
1 - MidiClock
2 - TeensyUsbMasterMidiClock
3 - TeensyUsbSlaveMidiClock
4 - TeensyUsbSlaveMidiClockMonitor
```
1-3 work with no issues , however example 4 is having some issues.

It seems to lock up with an initial BPM of 42 or higher. The tempo LED freezes solid upon receiving a MIDI start byte. Additionally, there is no display of the tempo value, "bpm"or the "playing" or "stopped" messages on the the OLED. It only shows "uClock" .

If you power on the Teensy LC and the tempo of the main clock is below 42 bpm, the tempo LED does not lock up and you can adjust the bpm beyond 42 bpm (20-300 BPM is what I tested from external sequencer clock) and it will continue to work. However, there is still no display of the tempo value, "bpm"or the "playing" or "stopped" messages on the the OLED.

Changing the #define `EXT_INTERVAL_BUFFER_SIZE` value doesn't seem to resolve the issue. Anything else you recommend? 

I'll keep poking around and see what I can find but I wanted to report in case you or someone else can reproduce. Thank you.

midilab, 2022-08-12:
Did your pull request solve this issue?

m-r-m-s, 2022-08-12:
Hey @midilab No - this issue came after as the code would not compile previously due to the volatile reference pointer. 

The current TeensyUsbSlaveMidiClockMonitor example code compiles and works but with the above issues. I have also tested this example with DIN MIDI instead of usbMIDI and the same issues occur.

midilab, 2022-08-13:
i will check asap, thanks for test and report!

m-r-m-s, 2022-08-13:
No problem and thank you!

m-r-m-s, 2022-08-13:
Hi @midilab , 

A quick update to this issue with more detail: The same issue occurs with the Xiao as well. I also forgot to mention that even though the tempo LED doesn't work after 42 bpm (testing using Abelton link from iOS) it does continue to still be responsive to Start and Stop messages and receiving Clock according to my MIDI Monitor.

I also tried both options for setting external sync mode and no difference:
` uClock.setMode(1)`
`uClock.setMode(uClock.EXTERNAL_CLOCK)`

Below is the `UsbSlaveMidiClockMonitor` sketch modified for the Xiao:

```
#include <Arduino.h>

/* USB MIDI Sync Slave Box Monitor
 *  
 * This example demonstrates how to create a
 * MIDI hid compilant slave clock box with 
 * monitor support using oled displays
 *
 * Making use of a 250 usceconds timer to
 * handle MIDI input to avoid jitter on clock
 * 
 * You need the following libraries to make it work
 * - u8g2
 * - uClock
 *
 * This example code is in the public domain.
 */
 
#include <U8x8lib.h>

#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;
 
// Create a new instance of the Arduino MIDI Library and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI_USB); // Type, SerialPort, Name

//
// BPM Clock support
//
#include <uClock.h>

U8X8 * u8x8;

// IntervalTimer teensyTimer;

// MIDI clock, start, stop, note on and note off byte definitions - based on MIDI 1.0 Standards.
#define MIDI_CLOCK 0xF8
#define MIDI_START 0xFA
#define MIDI_STOP  0xFC

char bpm_str[4];
float bpm = 126;
uint8_t bpm_blink_timer = 1;
uint8_t clock_state = 1;

void handle_bpm_led(uint32_t tick)
{
  // BPM led indicator
  if ( !(tick % (96)) || (tick == 1) ) {  // first compass step will flash longer
    bpm_blink_timer = 8;
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % (24)) ) {   // each quarter led on
    digitalWrite(LED_BUILTIN, HIGH);
  } else if ( !(tick % bpm_blink_timer) ) { // get led off
    digitalWrite(LED_BUILTIN, LOW);
    bpm_blink_timer = 1;
  }
}

// Internal clock handlers
void ClockOut96PPQN(uint32_t tick) {
  // Send MIDI_CLOCK to external gears
  MIDI_USB.sendClock(); 
  // MIDI_USB.sendClock(); // MIDI_USB.sendRealTime(midi::Clock);
  handle_bpm_led(tick);
}

void onClockStart() {
  MIDI_USB.sendStart();
  // MIDI_USB.sendStart(); //MIDI_USB.sendRealTime(midi::Start);
}

void onClockStop() {
  MIDI_USB.sendStop();
  // MIDI_USB.sendStop(); // MIDI_USB.sendRealTime(midi::Stop);
  digitalWrite(LED_BUILTIN, LOW);
}

// External clock handlers
void onExternalClock()
{
  uClock.clockMe();
}

void onExternalStart()
{
  uClock.start();
}

void onExternalStop()
{
  uClock.stop();
}

void setup() {
  // A led to count bpms
  pinMode(LED_BUILTIN, OUTPUT);

  //
  // MIDI setup
  //
  MIDI_USB.begin();
  MIDI_USB.setHandleClock(onExternalClock);
  MIDI_USB.setHandleStart(onExternalStart);
  MIDI_USB.setHandleStop(onExternalStop);

  //
  // OLED setup
  // Please check you oled model to correctly init
  //
  u8x8 = new U8X8_SH1106_128X64_NONAME_HW_I2C(U8X8_PIN_NONE);
  u8x8->begin();
  u8x8->setFont(u8x8_font_pressstart2p_r); 
  u8x8->clear();
  u8x8->drawUTF8(0, 0, "uClock"); 

  //
  // uClock Setup
  //
  // Setup our clock system
  uClock.init();
  uClock.setClock96PPQNOutput(ClockOut96PPQN);
  // For MIDI Sync Start and Stop
  uClock.setOnClockStartOutput(onClockStart);
  uClock.setOnClockStopOutput(onClockStop);
  
  // uClock.setMode(uClock.EXTERNAL_CLOCK); // is this same method as uClock.setMode(1) ?

  // Teensy specific
  // make use of 250us timer to handle midi input sync
  // teensyTimer.begin(handleMidiInput, 250); 
  // teensyTimer.priority(80);

  // set to external sync mode
  // NOTE: this is used by the example here: https://github.com/midilab/uClock/blob/master/examples/TeensyUsbSlaveMidiClock/TeensyUsbSlaveMidiClock.ino
  uClock.setMode(1);

  // wait until device mounted
  while( !USBDevice.mounted() ) delay(1);
}

void handleMidiInput() {
  while (MIDI_USB.read()) {
  }
}

void printBpm(float _bpm, uint8_t col, uint8_t line) {
    int b = (int)_bpm;
    int c = (int)((_bpm-b)*10);
    itoa(b, bpm_str, 10);
    if (b > 99) {
      u8x8->drawUTF8(col, line, bpm_str);
    } else {
      bpm_str[2] = '\0';
      u8x8->drawUTF8(col, line, " ");
      u8x8->drawUTF8(col+1, line, bpm_str);
    }
    u8x8->drawUTF8(col+3, line, ".");
    itoa(c, bpm_str, 10);
    u8x8->drawUTF8(col+4, line, bpm_str);
    u8x8->drawUTF8(col+5, line, "bpm"); 
}

void loop() {

  handleMidiInput();

  if (bpm != uClock.getTempo()) {
    bpm = uClock.getTempo();
    printBpm(bpm, 8, 7);
  }
  if (clock_state != uClock.state) { 
    clock_state = uClock.state;
    if (clock_state >= 1) {
      u8x8->drawUTF8(0, 7, "playing"); 
    } else { 
      u8x8->drawUTF8(0, 7, "stopped "); 
    }
  }
}
```

midilab, 2022-09-12:
i couldn't reproduce this bug.

I test using teensy LC/Teensyduino and ableton live 9 as master clock via usb using the default example for teensy usb monitor.

Starting with 20bpm and raising up doesn't trigger the bug for me.

Did you try to compile using arduino/teensyduino environment instead of platformio?

thanks for the tests and inputs so far.

midilab, 2022-09-21:
what are the version of uclock are you testing on?

midilab, 2022-09-21:
i've just release the last master changes you have pull for examples for 1.1.2 version

i will try to test again soon and report back.

midilab, 2022-09-21:
The library versions im using:
u8g2: 2.28.10

midilab, 2022-09-24:
so finaly i was able to reproduce the bug. i also find the root of the problem, i will come up with a solution soon...

once again thanks for testing out!

midilab, 2022-09-24:
the problem was on the printBpm() function on sketches of slave midi clock monitors examples.
i change the inplementation to avoid crashing at first clocks received.

so everything is on master branch, can you please test all the examples you have found bugged? if everything is ok i will release a new bugfix version for the library.

midilab, 2022-09-26:
have you tested the examples on the master branch? they are the ones modified.

midilab, 2022-09-28:
I've commited a new version of midi monitor sketches get ride off printBpm function and make use of String class to avoid problems like that.

Is that possible to do some test again with the new commited version?

On my side looks ok on teensy, but i didn't test the leonardo version.

midilab, 2022-09-29:
thanks for the tests so far.

since this is a opensource project i always encourage people to contribute. if you think there is any improvements(like this example support for 128x32 oleds), feel free to just pull the changes and then we can discuss the changes over the pull request(thats keeps a better tracking schema over repository).

midilab, 2022-09-29:
> 

yes you can change that, just a matter of handle String() parameters over numbers without decimal values.