midilab/uClock #23: TeensyUsbMasterMidiClock no name.c

Jackson-Devices, 2023-05-15:
"The custom name is in the "name.c" tab." there is no name.c tab when loading this project, please update this example.  Thanks

midilab, 2023-05-15:
good point! thanks for reviewing that! we can add this to that branch release as a fix too.

file: name.c
```c++
// To give your project a unique name, this code must be
// placed into a .c file (its own tab).  It can not be in
// a .cpp file or your main sketch (the .ino file).

#include "usb_names.h"

// Edit these lines to create your own name.  The length must
// match the number of characters in your custom name.

#define MIDI_NAME   {'u','c','l','o','c','k','_','1'}
#define MIDI_NAME_LEN  8

// Do not change this part.  This exact format is required by USB.

struct usb_string_descriptor_struct usb_string_product_name = {
        2 + MIDI_NAME_LEN * 2,
        3,
        MIDI_NAME
};```

midilab, 2023-05-15:
Fixed at [v1.3.0 release](https://github.com/midilab/uClock/releases/tag/v1.3.0)

closed 2023-05-15 by midilab