midilab/uClock #28: 'IntervalTimer' does not name a type - Seeed Xiao m0

vanzuiden, 2023-11-19:
Hi,

First of all, thank you for creating this library. I have been working on the prototype of a sync device and have been struggling to get a solid external midi sync going. From what I’ve seen your library seems to be a great starting point for what I’m trying to achieve.

I’m coming from a pi pico with python setup and just received a Xiao m0 today to test your library.

I had some trouble getting the library to run. I later found that version 1.3.0 or newer fails to compile due to the following error

```In file included from PATH/Arduino/libraries/uClock/src/uClock.cpp:46:0:

PATH/Arduino/libraries/uClock/src/platforms/samd.h:11:1: error: 'IntervalTimer' does not name a type
 IntervalTimer _uclockTimer;
 ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1```

Version 1.2.0 or older compiles and runs fine on the Xiao.

I'm looking forward to be able to build a prototype with uClock!

midilab, 2023-11-20:
hi @vanzuiden ,

Given that most open-source projects rely on other open-source projects, it's common practice to adhere to well-known working versions of each dependency library as a strict requirement for usage. I apologize for not having included versioning requirements notes for each platform in the documentation yet, which would help users identify the correct version to use. Thanks for your testing; we now know that version 1.2.0 is the latest working version for uClock. Therefore, we should stick with it until uClock receives an update to support the newer versions.

During my work on the initial support for the SAMD timer interval library in uClock, I encountered several bugs and struggled with inadequate documentation. Due to these challenges, I don't consider SAMD as an ideal platform for handling timers. However, it's worth noting that some users have reported successful usage of SAMD with uClock without encountering any issues. I recommend conducting thorough testing to measure timing accuracy and overall performance before deciding whether to proceed with SAMD or explore alternative platforms.

If you're looking for an affordable microcontroller, I recommend using an ESP32, which is more powerful and is one of the newest supported platforms for uClock. Alternatively, you can test your prototype against SAMD using version 1.2.0 and see if it fits your requirements.

Thank you for your understanding and collaboration.

Best regards,

vanzuiden, 2023-11-20:
Thank you for your reply @midilab! I appreciate your elaborate answer and the tip for the ESP32. I'll look into it. 

Is it possible to publish older versions to platformio? Since 1.2.0 is not available on platformio I had to switch to the Arduino IDE. This led to a bit of confusion starting out with platformio.

I'll start on my breadboard prototype based on the SAMD today and will properly test it. If you ever need someone to test any future updates I'm happy to help out.

Since the official Arduino support for the Pi Pico became available earlier this year, are you possible still looking to support these boards? I saw mention about this on an older closed issue. It is the most affordable microcontroller, quite powerful and is very easy to get a hold of. I'm also available to help test this!

Thank you again! 

All the best, Robbert

midilab, 2023-11-20:
wait, i think i missread your post, you mean 1.2.0 uclock version? or 1.2.0 samd libraries support?

If this is related to uClock version then maybe it is a problem related to the updates on 1.3.0, that could be easy to fix.

midilab, 2023-11-20:
but how did you test on platformIO? its published 1.4.0 +

vanzuiden, 2023-11-20:
I switched to the Arduino IDE mac app and tested the older versions.

vanzuiden, 2023-11-20:
Thank you @midilab for the swift replies and fix! I appreciate it a lot.

midilab, 2023-11-20:
you can try v1.5.1 now on platformIO

vanzuiden, 2023-11-20:
Hi, just had the time to test this out and v1.5.1 works as expected. Thank you!

closed 2023-11-21 by midilab