midilab/uClock #39: uClock does not compile with ESP32-C3

underwoodblog, 2024-04-29:
As the title says:

In file included from G:\PortableApps\arduino\portable\sketchbook\libraries\uClock\src\uClock.cpp:52:
G:\PortableApps\arduino\portable\sketchbook\libraries\uClock\src\platforms/esp32.h:28:48: error: macro "portYIELD_FROM_ISR" passed 1 arguments, but takes just 0
     portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
                                                ^
G:\PortableApps\arduino\portable\sketchbook\libraries\uClock\src\platforms/esp32.h: In function 'void handlerISR()':
G:\PortableApps\arduino\portable\sketchbook\libraries\uClock\src\platforms/esp32.h:28:5: error: 'portYIELD_FROM_ISR' was not declared in this scope
     portYIELD_FROM_ISR(xHigherPriorityTaskWoken);

old ESP32 and ESP32 S2 compiles, ESP32 C3 not!

macsbug, 2025-02-03:
A compilation error occurs on ESP32 S3.

To address this issue, I took the following steps:
--------------------------------------
uClock.cpp
#if defined(CONFIG_IDF_TARGET_ESP32)
      #include "platforms/esp32.h"
#endif

#if defined(CONFIG_IDF_TARGET_ESP32S3)
      #include "platforms/esp32S3.h"
#endif

--------------------------------------
Added esp32S3.h, which is a modified version of esp32S.h.

//#define TIMER_ID  0    // revision
#define TIMER_ID  1000000 // revision

  //_uclockTimer = timerBegin(TIMER_ID, 80, true); // revision
    _uclockTimer = timerBegin(TIMER_ID);           // revision

    // attach to generic uclock ISR
  //timerAttachInterrupt(_uclockTimer, &handlerISR, false); // revision
    timerAttachInterrupt(_uclockTimer, &handlerISR);        // revision

    // init clock tick time
    //timerAlarmWrite(_uclockTimer, init_clock, true); // revision

    // activate it!
  //timerAlarmEnable(_uclockTimer);                // revision
    timerAlarm(_uclockTimer, init_clock, true, 0); // revision
}

void setTimer(uint32_t us_interval)
{
  //timerAlarmWrite(_uclockTimer, us_interval, true); // revision
    timerAlarm(_uclockTimer, us_interval, true, 0);   // revision
}
--------------------------------------

midilab, 2025-02-03:
Hi @underwoodblog

Thank you for reporting this issue. Unfortunately, I was unable to reproduce the problem using the latest 3.11 ESP32 Core and the uClock library from the main branch.

Could you please provide details about your environment setup? Specifically:

- **Arduino IDE version**:
- **ESP32 module** (e.g., 3.0.5 or 2.0.x):
- **uClock version**:

This information will help me investigate the issue more effectively. Let me know if you need any clarification or assistance.

Best regards,

midilab, 2025-02-18:
closing due to inactivity

closed 2025-02-18 by midilab