midilab/uClock #40: Sync to external Eurorack trigger

carlosedp, 2024-07-22:
Hi, I'm writing an Eurorack module for clock generation and division and wonder if I can sync my module that uses uClock to an external trigger coming from another Eurorack module.

The input trigger is usually a square wave from another oscillator or LFO (not MIDI).

Can I use the uClock.clockMe() for this? Any tips?

Thanks!

midilab, 2024-07-23:
Hi @carlosedp,

Yes it is possible to sync to external clock. one of the uclock main features is the phase lock mechanism to keep a good and solid internal clock based on external one.

uClock.clockMe() should be called at each low or high state of you square wave depending of the rise/fall style of your clock signal.

Rigth now the expected clock is 24PPQN, we have a TODO list to include the option to change expected external clock too.

But for now if you have any external clock different from 24PPQN you need to handle it on your code. For example: if you have a 48PPQN input clock, you should trigger uclock.clockMe() each second tick.

A code example
```c++
uint8_t clockDiv48Trig = 0;
void ppqInterrupt() {
  if (clockDiv48Trig == 0 || clockDiv48Trig % 2 == 0) {
    uClock.clockMe();
  }
  ++clockDiv48Trig;
}         
```

and a thread about the subject:
https://github.com/midilab/uClock/issues/37

If you external clock is 24PPQN than you only need to call uClock.clockMe() each external signal rise or fall.

carlosedp, 2024-07-23:
Got it but I think in my case it's kinda the opposite... I need to generate a 24PPQN from a single pulse (can I call it 1 PPQN?)... right?
Is this viable or precise?

midilab, 2024-07-23:
i dont think it exist such of resolution 1PPQN. Most eurorack modules are based on 48PPQN or 24PPQN, if it is a single LFO there is no resolution at all, it is only a signal that you will try to sync to.

My suggestion is that you try it first uClock.clockMe() at each signal input LOW or HIGH(choose one of those only), and check it how it behaves, for clock generation.

Mabe you can start with the example:
https://github.com/midilab/uClock/blob/main/examples/GenericMasterOrExternalSync/GenericMasterOrExternalSync.ino

your clock output will be done inside onSync24Callback function based on bellow example(wich outputs at 24PPQN resolution)

The example is doing clockMe() inside loop() function, but ideally you should use interruption linked to your clock input PIN instead of doing inside loop().

houtson, 2024-09-22:
I think the OP is looking for a 'tap tempo' style of input, with a trigger per step/quarter note.

houtson, 2024-09-22:
> But for now if you have any external clock different from 24PPQN you need to handle it on your code. For example: if you have a 48PPQN input clock, you should trigger uclock.clockMe() each second tick.

Thanks for the 48PPQN example, I'll use that for syncing with old Korg DINSYNC synths.

Cheers Paul

carlosedp, 2024-11-04:
That's exactly what @houtson mentioned... suppose I want to sync my module that uses uClock with MI Marbles T2 output for example... it would output 120BPM in the middle position... if I connect it to my module's INPUT that triggers the clockMe function, it wouldn't work since it expects 24PPQN which would be 24x Marbles output.

Is there any way to make the clockMe set the input PPQN for the sync signal?

midilab, 2024-11-17:
Yes, you can use the code snippet on this thread adjusted to your context. At each 24 pulses from marbles it will triger 1 uClock.clockMe(). 

```c++
#define CLOCK_COUNTER 24
uint8_t clockDivTrig = 0;
void ppqInterrupt() {
  if (clockDivTrig == 0 || clockDivTrig % CLOCK_COUNTER == 0) {
    uClock.clockMe();
  }
  ++clockDivTrig;
}         
```

IC-Alchemy, 2024-12-10:
@carlosedp 

Marbles is a [confusing example](https://pichenettes.github.io/mutable-instruments-documentation/tech_notes/clock_processing_with_ramps/) and my answer assumes you were not asking about syncing MIDI to Marbles in particular but to any other eurorack module, like Grids for example.

> > I need to generate a 24PPQN from a single pulse (can I call it 1 PPQN?)... right? 

A single pulse is very different from 1 PPQN, although it took me a while to finally understand all this, and I have a background in music and degree in jazz performance. 

  **24 PPQN** describes timing resolution used for MIDI clock.

I think of PPQN like Frames Per Second (FPS)

FPS is the number of individual images (frames) displayed each second.
PPQN is the number of pulses used to represent one beat (specifically, a quarter note) in music

**Higher PPQN** means the musical timing is divided into more parts, allowing for finer placement of notes and rhythms.

For example Ableton uses 960PPQN which is a high enough resolution where you can come close to any swing imaginable.

carlosedp, 2024-12-10:
Yea, I know... :) When I mean 1PPQN I mean imagine syncing the library to an LFO with a square wave outputting pulses every second (60BPM) ... the idea is that while pretty unstable, this pulse train would drive the clock at 60BPM too.

I think Pamela's workout has a setting where you can set the input PPQN for the output clock where the default is 24PPQN but it can go as low as 1PPQN.

Check this part of the video: https://youtu.be/mkoVfQq7m8Y?t=285

IC-Alchemy, 2024-12-14:
I've made tons of devices that move forward one step on each rising edge trigger.
This can be done with out any clock at all, and without keeping track of time.

```
If (risingEdge)
    step++; 
```
For this to be synced with uClock, 
so that you could output a midi clock that is synced with your LFO for example,
 you'd have to decide what note value you want to assign to each step.

This won't work if you think of an LFO as being 1PPQN 

The standard sequencer defaults to 16th notes, like on a TB303.

Each 16th note consists of 6PPQN, assuming your clock is MIDI at 24PPQN

midilab, 2025-02-16:
Hi @carlosedp, @houtson, @IC-Alchemy, and @doctea,

I've added an experimental feature that allows setting the input sync clock for different PPQN resolutions. This is currently untested, so I'm reaching out to gather feedback from you all.

The feature is implemented using setClockPPQN(PPQNResolution resolution) method. 

```cpp
uClock.setClockPPQN(PPQN_4);
```

This sets the resolution to 4 PPQN (Pulses Per Quarter Note), meaning each pulse corresponds to one step. For reference, here are the available PPQN resolutions in an extended list:

```cpp
enum PPQNResolution {
    PPQN_4 = 4,
    PPQN_8 = 8,
    PPQN_12 = 12,
    PPQN_24 = 24,
    PPQN_48 = 48,
    PPQN_96 = 96,
    PPQN_384 = 384,
    PPQN_480 = 480,
    PPQN_960 = 960
};
```

Please note that while this code compiles and the logic appears sound on paper, it hasn't been thoroughly tested yet. I'm especially interested in hearing from anyone who has tried using this feature or encountered any issues with it.

This way we can have higher resolution clock running internally(good for groove/shuffle stuff) and and keep on sync with external lower resolution clocks.

Thank you in advance for your feedback and insights.

Best regards,

carlosedp, 2025-02-17:
That's awesome! I'm currently on vacation and will return next week, then I will take a look and play with it!
Thanks! Seems promising!

midilab, 2025-03-26:
The latest version on the develop branch is now ready for external sync with 1 PPQN and 2 PPQN. I’ve done some basic tests, but I don’t have a modular to test it with. If someone could please test the code to validate it for a new release, that would be great.

midilab, 2025-03-26:
@doctea 

This lower PPQN sync support, 1 PPQN in special, opens the door to implementing the tap function. I will commit a version with tap() support soon.

midilab, 2025-03-26:
I was also thinking about changing the names of the methods to avoid confusion:

* uClock.setClockPPQN() to uClock.setInputClockPPQN()
* uClock.setPPQN() to uClock.setOutputPPQN()

I'll probably make these changes before the official release. Right now, we'll keep it as it is.

doctea, 2025-04-09:
Hacked in CV clock input support (in an ugly fashion) into my project and got this roughly working using the `develop` branch and `PPQN_4`. Not sure if it is working like intended though -- getting very variable tempo even when fed a stable clock, and also getting the clock running at erratic speeds when the CV input is disconnected entirely.  That could be my hardware or surrounding project code at fault, tho -- I'll have to do some more experiments.

Question: is this intended to drive output ticks at 24 PPQN in a smooth and steady fashion, ie interpolating between input pulses?

midilab, 2025-04-09:
> Question: is this intended to drive output ticks at 24 PPQN in a smooth and steady fashion, ie interpolating between input pulses?

Yes, in that case you now have new Sync callbacks to make use wich will be independent from Input and Output Clock.

example:

```c++
uClock.setInputClock(uClock.PPQN_4); // the resolution you are expecting to receive in case uClock set to slave mode
uClock.setOutputClock(uClock.PPQN_96); // your sequencer internal/output clock cloud be any greater or equal to InputClock.
uClock.setOnSync48(yourCallbackAt48Ppqn); // send Korg DIN Sync48 to LINN LM-2, ARTURIA, ELEKTRON...
uClock.setOnSync24(yourCallbackAt24Ppqn); // send MIDI CLOCK
uClock.setOnSync4(yourCallbackAt4Ppqn); // send clock to ARTURIA , DOEPFER MCV 24 – MSY2 or 1/16th , Korg SQ-1 analog sync , most analog sequencers, Orthogonal Devices ER-101, TipTop Trigger Riot & Circadian Rhythm. Erica Synths Drum Sequencer , 010 Black Box (clock in)  Quite common to find as the default for Eurorack modules.
uClock.setOnSync2(yourCallbackAt2Ppqn); // send clock to KORG (SQ-1), Volcas, Teenage Engineering Pocket Operators (POs sync on audio pulses, essentially a click track).
uClock.setOnSync1(yourCallbackAt1Ppqn); // send clock to DOEPFER DARK TIME, Korg SQ-10, Zularic Repetitor, (Noise Engineering), Erica Pico Seq & Pico trigger , Delptronics Triggerman, Disting Mk3 & 4, Modcan Touch Sequencer.

```

For the other strange behaviours you mentioned, can you please open a new issue since this one is already closed as enhancment done.

carlosedp, 2025-04-09:
Thanks @midilab ... since I wrote a bunch of the logic myself, I'll try to test and start porting the clocking core for my module to uClock.

I also agree on renaming the functions to set the input/output ppqn.

I'll report soon... thanks a lot, this looks neat!

midilab, 2025-05-22:
Implemented at [Release 2.2.1](https://github.com/midilab/uClock/releases/tag/v2.2.1)

closed 2025-05-22 by midilab