midilab/uClock #43: Workaround for strange MIDI delays using RP2040 and the MIDI library on the hardware serial ports

ichingxs4all, 2024-09-24:
I recently found a workaround for strange irregular delays in outputting MIDI clock through the hardware serial port on a RP2040 based boards like the Pico or the Waveshare RP2040 Zero.
I can pinpoint it to every read operation that is executed in the main loop lik : void loop(){ MIDI.read()} . Or a direct serial port read without the MIDI library.
USB MIDI does not have this problem when it is used solo, but both together with an hardware serial device it also this strange behaviour. The result is a BPM that is very irregular and way below the BPM you set with setTempo.

So the workaround I found after al lot of experimentation is using the dual core mode of the RP2040. This solves the problem.

// Running on core1
void setup1() {
// Initialize UART midi stack
  MIDI.begin(MIDI_CHANNEL_OMNI);
  setHWMidiHandlers();
  MIDI_USB.begin(MIDI_CHANNEL_OMNI);
  setMIDI_USBHandlers();
}

void loop1() {
  MIDI.read();
  MIDI_USB.read();
}

doctea, 2024-10-16:
Do you have a simple example that demonstrates this, or can you post your full code?  I output serial MIDI clock in my RP2040+uClock project, and I don't think I've noticed anything strange like this so far...

midilab, 2025-02-03:
Closing due to inactivity...

closed 2025-02-03 by midilab