midilab/uClock #49: Feature request(?): pause/continue clock

doctea, 2025-03-26:
Hi!  Maybe I'm looking in the wrong place but I couldn't see that it is possible to do this with the current version of the library?

I.e., to pause playing and then to continue where we left off.

Am I missing that this is already implemented somehow?

If not, here is a quick attempt at implementing this.  It seems to be working OK in my project so far :)

https://github.com/doctea/uClock/tree/feature-pause-continue

I'll make a PR if this would be useful and if you think its the right way to go about it.

Thanks,

midilab, 2025-03-26:
Indeed, there is no real pause in the current version.

However, one question comes to mind:

What kind of pause do we want when we are in slave clock mode? Should it be the type that pauses the internal tick state but continues to track the external tick location(when un-pause the internal tick continue from the same place as external tick)? Or should it pause the internal tick and resume from the same tick when unpaused(when un-pause it just continue the tick from where it stops)?

I've seen sequencers doing one or the other, and a few can do both through configuration.

I think an option to configure this via the API for the type of pause would be a good idea.

doctea, 2025-03-27:
> What kind of pause do we want when we are in slave clock mode? Should it be the type that pauses the internal tick state but continues to track the external tick location(when un-pause the internal [...]

I'm not sure I follow what you mean here.. 

Are the options,
a) still track the external ticks so that when un-paused we are still in sync with the external clock, but 'mute' any uClock-driven sequencer events that occur in the meantime?
b) complete pause and resume 'as if nothing had happened'?

.. or something else I'm not considering?

For my use cases I'd want "b".  "a" for me would be more like a 'mute sequencer' type of function that's distinct from pausing.  But maybe I'm confused?

> I've seen sequencers doing one or the other, and a few can do both through configuration.
> 
> I think an option to configure this via the API for the type of pause would be a good idea.

Definitely, if it is a choice between two behaviours then making it an option would be for the best :)

PascalMoonify, 2025-04-02:
Hi, it would be great. I would also suggest to add a new setOnClockContinue(callback) to declare a callback to manage that specific situation

doctea, 2025-04-11:
>I would also suggest to add a new setOnClockContinue(callback) to declare a callback to manage that specific situation

Added this (and setOnClockPause), although haven't tested.  Also opened a pull request https://github.com/midilab/uClock/pull/51

doctea, 2025-04-11:
> Should it be the type that pauses the internal tick state but continues to track the external tick location(when un-pause the internal tick continue from the same place as external tick)? Or should [...]

I think I understand better what you mean now @midilab :).  But I'm sorry that I still don't know what the correct answer is!

midilab, 2025-04-12:
> I think I understand better what you mean now @midilab :). But I'm sorry that I still don't know what the correct answer is!

Just throwing out a bit of a brainstorm question—there's no right answer here. If it’s worth implementing both approaches, I think that would be ideal.

I’ve always liked using classic and well-regarded machines as reference points when designing features—that's why I’m asking. You've probably worked with different setups and gear than I have, so it’s always good to raise these kinds of questions and compare notes.

midilab, 2025-07-27:
Implemented at 8a4b45cf901f4fac6ddad6d782da81d2f0614a31

closed 2025-07-27 by midilab