midilab/uClock #50: Testing external CV clock

doctea, 2025-04-09:
As requested in #40 , here's a separate issue to discuss weird behaviour :)

I've played around with it some more but I think my project hardware and software is not reliable enough to tell whether this problem is with the library or not.  Its much more likely to be deficits in my hardware and software.  

There's a lot going on in my project (SPI screen, SPI bidirectional GPIO, i2c ADC+DACs...) so its really not much surprise to me that I'm not getting solidly timed results from polling :).  I'm using multiple MCP23S17s as both inputs and outputs, and using one pin as the clock input test, and haven't had any luck getting interrupts working when a MCP23S17 pin changes.  So, the time elapsed between each recognised pulse varies wildly, and is almost certainly the source of the tempo fluctuations I'm seeing.   I think if I could get interrupts working then this would improve a lot, but I'll likely need a new hardware revision (or to make some modifications to my existing prototypes..) to prove that.

What is less clear to me is: should the uClock clock continue when the external pulses stop?  It probably works quite well as a 'tap tempo' system that the clock continues running, but if it is supposed to stay slaved to the external clock then I would expect it to stop once the pulse is removed.

Hope this makes some sense and is in some way helpful :)

doctea, 2025-04-10:
> However, when dealing with a gate signal via a multiplexer, things become a bit trickier. We can't buffer the incoming gate signal in the same way, so we can't guarantee that the signal will still be available to read at the next 250µs check.

Good news! I just quickly hacked together my code so that I check the MCP23S17's pin from an interrupt every 250us, setting a "cv_clock_ticked_flag" as appropriate.  Then I check that flag from my main loop(), calling uClock.clockMe() if appropriate, and.. SUCCESS!  

I'm now clocking my project from a square wave LFO at 4 PPQN.  Tempo sounds audibly solid, no noticeable jitter at all, and uClock's tempo display wavers by less than 1bpm -- seemingly more solid than when clocking from polled USB MIDI.  (I'll have to experiment with this 'checking MIDI clocks from interrupt' and see if that solves the jittery USB MIDI, too).

I'll continue exploring this when I get chance, but this is very heartening :).  It seems that on my platform at least that it is perfectly feasible to run all these 'heavy' features and also to process external clock in good enough time to be perfectly usable.

I'll have a think about the 'when should the internal clock know to stop if external sync is not found' thing, too -- my first thoughts are that if the external clock has not been detected by the time that it would be expected then the internal clock should stop dead too and then only continue playing when a new pulse is received.  Would that perhaps mean it would never get more than "PPQN" pulses ahead of the external clock, and should stay in phase with the external clock too...?

doctea, 2025-04-11:
>It’s possible that, depending on the master clock's timing, we could still miss a tick within that 250µs window 

I did consider this -- I'll have to do the maths to work out what BPM would be too fast to accurately clock or where pulses might start getting missed or combined.

I was also thinking that in order to avoid the possibility of multiple pulses being received before the main loop gets chance to service them, I could track the number of pulses received and clockMe() the correct number of times to ensure sync - although that would still throw the tempo estimation off.

>Do you know of any sequencer that does handle start/stop purely based on the clock signal? If so, it could be a great reference point for implementing this feature ourselves.

Hmm, well, for modular sequencers (that aren't effectively multiplying the clock to generate higher PPQNs), they don't necessarily have a concept of 'playing or stopped' and they just wouldn't advance at all if a pulse wasn't received.  So an option to emulate something close to that would be good.

I sometimes see clock multiplier modules and wonder how they deal with generating the extra pulses and unexpected missing pulses.. but as I don't own any and generate all my clocks from my own project using your wonderful library, I've never had the opportunity to really find out!

>That said, since you're already experiencing such a solid lock with the master clock, I think this is totally usable in real-world situations. 

I did note that it seems to take some time for the uClock displayed tempo to catch up with the real pulses, so hugely varying external clocks might cause things to go a bit haywire.  I'll have to set up a patch to test and tell if it ever actually loses sync and investigate mitigations if so.

midilab, 2025-04-11:
> I did note that it seems to take some time for the uClock displayed tempo to catch up with the real pulses, so hugely varying external clocks might cause things to go a bit haywire. I'll have to set up a patch to test and tell if it ever actually loses sync and investigate mitigations if so.

Using uClock.getTempo() will return the current BPM at the time it's called. As far as I know, no clock source is completely free from micro-variations, jitter, or timing inconsistencies—especially when the clock comes from a digital source, where a lot happens between internal processing and the actual clock tick firing.

Because of that, I implemented a buffer to calculate an average BPM reading in slave mode. Currently, in the develop branch, the buffer size has been increased to 128. This is what creates the sensation of the tempo taking a moment to "sync." However, this delay is only in the displayed value—for display purposes only. Internally, the clock is already in sync. The perceived lag comes from averaging over a larger buffer, not from the actual clock timing.

You can change the buffer size to 24, which was the default value in the previous uClock release. You can even try setting the buffer size to 1—that way, you’ll clearly see the BPM "display" variations I’m talking about. It’s a great way to understand how much fluctuation is actually present in real-time clock signals.

```c++
// for smooth slave tempo calculate display you should raise this value 
// in between 64 to 128.
// note: this doesn't impact on sync time, only display time getTempo()
// if you dont want to use it, set it to 1 for memory save
#define EXT_INTERVAL_BUFFER_SIZE 128
```

> Hmm, well, for modular sequencers (that aren't effectively multiplying the clock to generate higher PPQNs), they don't necessarily have a concept of 'playing or stopped' and they just wouldn't [...]

I’ll think about a way to create a proper test for this. The main issue arises when the internal clock is running at a much higher rate than the input clock. In that case, there are still a lot of internal ticks queued up to be generated—or not—depending on whether the input clock has actually stopped. This can create ambiguity and potential timing issues.

> I did consider this -- I'll have to do the maths to work out what BPM would be too fast to accurately clock or where pulses might start getting missed or combined.

I was also thinking that in order to avoid the possibility of multiple pulses being received before the main loop gets chance to service them, I could track the number of pulses received and clockMe() the correct number of times to ensure sync - although that would still throw the tempo estimation off.

Let me know, and we can properly document this setup so others can easily use it in modular environments.

midilab, 2025-04-11:
Also you can check this great read about clock, jitter and latency, also some notorious machines and their clock jittering.

https://www.innerclocksystems.com/litmus

midilab, 2025-05-22:
Closing due to inactivity

closed 2025-05-22 by midilab