midilab/uClock #53: uClock update 2.2.1 does not play nice with Adafruit SSD1306 libraries

awonak, 2025-05-26:
I recently upgraded to the new v2.2.1 library (very excited for the PPQN_4 external clock!) and unfortunately it appears that `uClock` and `Adafruit_SSD1306.h`/ `Adafruit_GFX.h` do not play nice together.

Just by importing the `uClock` library, the display ceases to draw to the screen. 

Here I have a basic working blink sketch that renders the text "uClock" to the screen during startup and implements a simple software blink function.

https://gist.github.com/awonak/90b1af635b89092cbd664d3c79fbe15b

In that sketch if you uncomment line 12 which imports `<uClock.h>` (verified with the latest HEAD commit from GitHub 35074971a32d958c650088f5df21670236f89cb2) you will notice that the display no longer renders our text. Aside from that, the software blink function still works as expected.

Further expanding on this, we can start to implement basic `uClock` functionality from the repository example code.

https://gist.github.com/awonak/216033906de08a8c342b1d592edcf1d0

When the script starts, both our existing blink and our newly implemented `uClock` callback start working as expected. The display still does not work. After 5 seconds, we call a display function with a few different tests.

```cpp
void updateDisplay() {
    display.clearDisplay();
    display.setCursor(10, 10);

    // With either of these line uncommented, the software blink stops when this function is called.
    display.print(F("Hello"));
    display.print(123);

    // If this line is uncommended, the uClock callback is executed significantly faster from initial 
    // execution, and the software blink stops when this function is called.
    display.print("Hello");

    // With this line uncommended, the firmware fails to start.
    display.print(uClock.getTempo());
    
    display.display();
}
```

If I checkout branch v2.1.0 (c0196f31a288c48d0a6e5fde641981881fc0b0ba), then the above test script executes without any issues (after reverting `setClockMode` and `setOnOutputPPQN` to their original method names)

awonak, 2025-05-26:
I also write a quick test using the u8g2 library in page buffer mode, and that display library is working as expected:
https://gist.github.com/awonak/20e8144c483eff4384b6a10b4e0423f6

awonak, 2025-05-26:
I tried to write a test sketch using `uClock` with `u8g2` library in Full Screen Buffer mode, but the library takes up about 78% of the Nano's dynamic memory (I think the uClock takes up about 32%), so I wasn't even able to test that configuration.

```
Sketch uses 14872 bytes (48%) of program storage space. Maximum is 30720 bytes.
Global variables use 2238 bytes (109%) of dynamic memory, leaving -190 bytes for local variables. Maximum is 2048 bytes.
```

Looking back at the previous test with the `uClock` and `Adafruit_SSD1306.h`/`Adafruit_GFX.h` libraries, that test sketch uses a total of 49% of dynamic memory, so I'm not sure that a lack of dynamic memory is the issue.

```
Sketch uses 17930 bytes (58%) of program storage space. Maximum is 30720 bytes.
Global variables use 1013 bytes (49%) of dynamic memory, leaving 1035 bytes for local variables. Maximum is 2048 bytes.
```

midilab, 2025-05-26:
Great to hear that it works!
If you're not in a hurry to release, I can create a method to set the buffer via the API and include it in a new release — something like uClock.setExtIntervalBuffer();.

midilab, 2025-05-27:
The only recommendations is no dynamic allocation after start using uClock, i would make uClock.init() mandatory only after all the setup is done, so dynamic allocate memory once and forever, the api should take care of it. 

I would like to avoid flags because the library is intent to people that are not programmers or tech expert, something that can work without doing much for musicians and creative people to use without hard times to setup.

awonak, 2025-05-27:
Excellent, thanks for the quick fix! I'll test this out today or tomorrow and report back.

awonak, 2025-05-28:
I did a few quick tests, and I can confirm that this works as expected.

### Happy path test cases  ✅

In this case we initialize with the default value of 1.
```cpp
# uClock::init
if (ext_interval_buffer == nullptr)
        setExtIntervalBuffer(1);

# Test Sketch
// Initialize the clock library without providing buffer size.
uClock.init();
```
In this case I set the default to  a very high amount and called `setExtIntervalBuffer()` with a small known working value.
```cpp
# uClock::init
if (ext_interval_buffer == nullptr)
        setExtIntervalBuffer(254);

# Test Sketch
// Initialize the clock library with a user provided buffer size.
uClock.setExtIntervalBuffer(4);
uClock.init();
```
In this test case, I proved that providing a max value _after_ calling init() does not modify the buffer size. (we should make sure this detail is explicitly clear in the documentation)
```cpp
# uClock::init
if (ext_interval_buffer == nullptr)
        setExtIntervalBuffer(1);

# Test Sketch
// Initialize the clock library with a very large user provided buffer size.
uClock.init();
uClock.setExtIntervalBuffer(254);
```

### Sad path test cases ❌

In this case I set the default buffer size to max and did not call `setExtIntervalBuffer()` with a lower known working value. The firmware froze on boot as expected.
```cpp
# uClock::init
if (ext_interval_buffer == nullptr)
        setExtIntervalBuffer(254);

# Test Sketch
// Initialize the clock library without providing buffer size.
uClock.init();
```
Conversely, we set the default to a low value and called `setExtIntervalBuffer()` with the max value. The firmware froze on boot as expected.
```cpp
# uClock::init
if (ext_interval_buffer == nullptr)
        setExtIntervalBuffer(1);

# Test Sketch
// Initialize the clock library with a very large user provided buffer size.
uClock.setExtIntervalBuffer(254);
uClock.init();
```

closed 2025-05-28 by midilab