midilab/uClock #56: error: 'class umodular::clock::uClockClass' has no member named 'setPPQN'

stevehunt100, 2025-06-10:
Compiler errors since uClock 2.2.1

I have been using uClock for years with no problems.  2.1.0 is fine.  Since 2.2.1 I get the following errors:

F:\Documents\Arduino\LFE\clocktest1\clocktest1.ino: In function 'void setup()':
F:\Documents\Arduino\LFE\clocktest1\clocktest1.ino:7:10: error: 'class umodular::clock::uClockClass' has no member named 'setPPQN'
    7 |   uClock.setPPQN(uClock.PPQN_24); // added from v2.0
      |          ^~~~~~~
F:\Documents\Arduino\LFE\clocktest1\clocktest1.ino:8:10: error: 'class umodular::clock::uClockClass' has no member named 'setOnPPQN'
    8 |   uClock.setOnPPQN(ClockOut96PPQN);  // was uClock.setClock96PPQNOutput(ClockOut96PPQN); after v2.0 uClock.h ,needs these 3 callBacks renaming
      |          ^~~~~~~~~
exit status 1

Compilation error: 'class umodular::clock::uClockClass' has no member named 'setPPQN'

simple sketch:

#include <uClock.h>  //https://github.com/midilab/uClock

void setup() { 
 // Set the callback function for the clock output to send MIDI Sync message.
  uClock.setPPQN(uClock.PPQN_24); 
  uClock.setOnPPQN(ClockOut96PPQN); 
  uClock.setOnClockStart(onClockStart);
  uClock.setOnClockStop(onClockStop);
  uClock.init();
}

void loop() { 
}

// The callback function which will be called by Clock each Pulse of 96PPQN clock resolution.
void ClockOut96PPQN(uint32_t tick) {    //  uClock.h
}
// The callback function which will be called when clock starts by using Clock.start() method.
void onClockStart() {
}
// The callback function which will be called when clock stops by using Clock.stop() method.
void onClockStop() {
}

Even your own example sketches have similar errors.

I am using Arduino IDE 2.3.5 for Teensy 4.1

Thanks

midilab, 2025-06-10:
Hi @stevehunt100,

All the breakchanges from >= 2.0 it is currently updated at [README](https://github.com/midilab/uClock?tab=readme-ov-file#uclock-v20-breaking-changes)

setClock96PPQNOutput(onClock96PPQNOutputCallback) is now renamed to setOnOutputPPQN(onOutputPPQNCallback), and its tick count is based on the PPQN setup using setOutputPPQN(clockOutputPPQNResolution). For clock ticks, you now use a separated callback via setOnSyncXX(onSyncXXCallback), where XX represents one of the available PPQN values.

Also you can use the [Interface](https://github.com/midilab/uClock?tab=readme-ov-file#interface) as base for the latest release.

Soon i will try to release a more complete documentation, just waiting the develop branch to come into main for lots of new features and the breakchanges needed to expand the features and api access to them.

stevehunt100, 2025-06-11:
Can you answer my question please?  I don't care about setClock96PPQNOutput()  I@M NOT USING IT!

I am using setPPQN which doesn't work now.

closed 2025-06-11 by midilab